makerbase-mks/MKS-TFT #107: z-breakfunction not working?

JONTY44, 2018-02-01:
why z axis breakpoint function does not work in autobed leveling, i mean when i tap continew button from the tft, print will start from the same z perameters where it stops ,but the autobed leveling feature is getting off..thats why i m getting leveling issues..why is it so?

gomines, 2018-02-01:
I’ve encounter the same problem in the “power off-resume” feature. 
Haven’t tried it, but i think activating the option on the motherboard’s firmware to save the grid on memory will make it load when you resume printing

JONTY44, 2018-02-03:
thats right but there should be load command added on continue button on tft, otherwise evenif the grid stored in firmware memory, it will not load at restart, so the problem continues

makerbase-mks, 2018-02-03:
@JONTY44  You need to preheat the printer. When the temperature reaches the target temperature, move the z-axis to the height you want to Continue. Click the "Continue" button.

gomines, 2018-02-07:
Any update on how to load the grid when restarting a print? 
I tried using this commands on the starting scripts and using the "more" buttons but with no success.
M374: Save grid to /sd/cartesian.grid  
M375: Load the grid from /sd/cartesian.grid and enable compensation (NOTE you must home after this)  
M500: Saves the probe points

JONTY44, 2018-05-14:
@makerbase-mks i know the continue button function, but what i m saying is that i m using autobed leveling feature and after pressing continue button the print will start but autobed leveling is getting off, so my print start on unlaveled bed, in other words my leveling points are not recovered.

suggestion:
i think u need to add M420 S1 GCODE command on top of continue button command so that  autobed level mesh is getting recovered before it continue from breakpoint. so that after continue button the print will follow autobed level mesh

closed 2020-06-06 by makerbase-mks