makerbase-mks/MKS-TFT #110: MKS cant command board and pictures with error

dudisr, 2018-02-03:
Hi  
Im trying to connect MKS TFT 2.8 V1.2 with ramps. 
Well  even though  using the latest SW 3.01. of MKS board cannot communicate with  ramps. In ramps wire is connected in AUX-1 with  red wire to the same direction of power supply. Im using a diff power supply for MKS. 

Other than that after updating to the latest SW of MKS, images are with error, but all  menu can be accessible.
Communication is 2500,  for all  boards. And i can command Ramps by pc, but no by touch screen. 

Anyone have a hunch ?

makerbase-mks, 2018-02-09:
1,you need to determine whether the baud rate is the same
2, can you control the other motherboard using TFT28?

dudisr, 2018-02-09:
The boards are brand new,  and as i  said by computer i  can command ramps but not by the MKS. Also with MKS i can receive the temperature value, but still  cannot command motors for ex. 
All  boards are using same baud rate 2500

Regarding the problem with  images i could fix. i  saw someone talking to  use in Windows the transfer to SD card,  with  that the images are appearing correctly

dudisr, 2018-02-09:
i  found an old case here that solved my  problem [https://github.com/makerbase-mks/MKS-TFT/issues/11](url) 
the problem was not with different baud rate on boards but i  changed the following parameters to work 

#########Printer type setting############################
#mainboard firmware setting(marlin:1; repetier:2; smoothie:3)
cfg_firmware_type:1 ( I have changed this to 3)
#machine setting (Normal:1; Delta:2)
cfg_machine_type:1
#baud rate setting(1:9600;2:57600;3:115200;4:250000)
cfg_baud_rate:4 (I have changed this to 3

dudisr, 2018-02-13:
Well  turns out that the solution presented before was not fully solved. 
At this moment i  can control motors and do the reading of temperature of sensors, but i can not trigger heated bed to preheat neither extruder. 
I will do more tests with MKS config file, but if anyone had anything like or know how to solve please let me know. @makerbase-mks

dudisr, 2018-02-17:
i did another tests changing the #mainboard firmware setting but nothing have changed. Actually i couldnt go  back  on the previous status ( i mean working motors and reading temp,  but no working extruder and heated bed). Although i  notice that if i change only for the right baud rate if i olnly turn on the boards it didnt work, but if if make the connection through Printrun by computer, the temperature suddenly starts to work on MKS. As i said before by computer it works but not from MKS. On the other hand if the board is turned off, MKS cannot read temperatures anymore. 
Im kind of stuck. @makerbase-mks  do you have any suggestion ? please answer