makerbase-mks/MKS-TFT #112: MKS PWC (Auto-Off) + SBase

gomines, 2018-02-10:
Hi, i was wondering how to run the:  PWC module V1.0+MKS Sbase V1.3 (+smoothieware)+MKS tft32. (i undestand it works with Smoothieware too).

I've done the following connections: 
[image]
(Using the B4 pin on the TFT, not the B0)

and added the following lines to the smoothieware's config.txt file:
switch.psu.enable                            true             # turn atx on/off
switch.psu.input_on_command       M80              # Power On
switch.psu.input_off_command       M81              # Power Off
switch.psu.output_pin                     1.22             # Open drain, inverted to control an ATX PSON signal
switch.psu.output_type                   digital          # on/off only</code>

With all of this, the problem is: For as long as i press the button, the printer is ON. But when i let go of the button the printer turns OFF. I think the 1.22 pin does not send the right signal to the relay to keep the printer ON.

Any suggestions? Does this module really work with the Sbase?

makerbase-mks, 2018-02-23:
switch.psu.enable                            true             # turn atx on/off
switch.psu.input_on_command                  M80              #
switch.psu.input_off_command                 M81              #
switch.psu.output_pin                        1.22          # open drain, inverted
switch.psu.output_type                       digital          # on/off only
switch.psu.failsafe_set_to                   0                # so the ATX turns off on a system crash
#switch.psu.ignore_on_halt                   false
switch.psu.startup_state             true

gomines, 2018-02-23:
Still not working :( could it be the firmware? or a faulty relay?

makerbase-mks, 2018-02-23:
Can you test whether P1.22 has output high level?

gomines, 2018-02-28:
How can I test this? and can you explain a little more at which level it is supposed to be?

makerbase-mks, 2018-03-02:
Do not release the button and use a voltmeter to measure whether the P1.22 pin outputs a high point

makerbase-mks, 2018-03-02:
Do not release the button, use a voltmeter to measure P1.22 pin is high

KuroiSabato, 2018-03-02:
little question:
want to use bigger  push button (19mm), but it is a 24volts.
if i connect it to the 24v power supplier, should it work? Or I do need a 240v button and connect it to the pwc as the diagram?

zheka1980, 2018-07-02:
I try to connect MKS TFT35 and MKS PWC 2.0
 have found instrution - PB4 - is auto power off (right upper onnector on PWC), and PB0 - is power detection (right down onnector on PWC). I have connected, how it written. When ON button pressed - printer works. When I press button agait to make printer OFF - printer was ON still, without any reaction. It not catching POWER LOSS signal.

 I tried to change connection - if I connect PB4 to VCC - I see on display "closing machine". But on PB0 - still +5v. AND MKS PWC not take off power. I think than some ouptup pin on MKS TFT32 must go to logic "0" after power loss and appearing "closing machine" after POWER LOSS ?

 Anybody have MKS TFT35 and MKS PWC?

zheka1980, 2018-07-02:
And I tried to add to config text 

switch.psu.enable true # turn atx on/off
 switch.psu.input_on_command M80 #
 switch.psu.input_off_command M81 #
 switch.psu.output_pin 1.22 # open drain, inverted
 switch.psu.output_type digital # on/off only
 switch.psu.failsafe_set_to 0 # so the ATX turns off on a system crash
 #switch.psu.ignore_on_halt false
 switch.psu.startup_state true

But p1.22 is always ON.

gomines, 2018-07-02:
Hi, i started this issue cause i couldn't make my PWC work, had the same problem as you mention. Had the connections as i wrote before and did everything as the manual did, even looked everywhere on the internet (also saw some russian posts) and after all of that my pwc still didn't work!

Changed my Sbase to an Gen v1.4 cause i wanted to use different drivers. In this case using marlin i was confident that the PWC would work after reading issue #59 and again, my PWC didn't work.

Don't know if the module has an issue or what, but i'll show you an alternative and is what i'm currently using on my 3D printer: the Arduino module Relay. This module is very easy to connect and started working from my first try.
[image]

I know this ain't a solution to make your PWC work, but sincerely i've given up on this module to work and there ain't further information about it. I'll leave you with the information for connections on the Arduino Relay module that i used:
http://3dtoday.ru/blogs/sergey1232/modernization-of-power-supply-flyingbear-p902-automatic-turnoff-after-/

gomines, 2018-07-02:
After working with this module, i was thinking if having the push button connected to all 3 cables: Common, NO (normally open) and NC (normally closed) should have something to do with the problem... since in this arduino module you only need to have C and NO to make it work.

nip2k2, 2018-07-15:
Hi,
I have MKS TFT32 + MKS GEN v1.4 and MKS PWC with Marlin. Really dont know how to make it work...
- I´ve connect like the diagram.
- I´ve make modificacations on mks_config.txt
- I´ve make modificacations on configuration.h and pins.RAMPS.h (Marlin).
Very frustrating!
Printer power goes off if I release power button...
Makerbase-mks any solution?
Thanks.

nip2k2, 2018-07-15:
Hi!! I´ve try everithing including not connecting all 3 wires from the push buttom, I think its a problem with the button.... unbelievable, cant make it work!
Makerbase-mks??????
Please!

hermitcrabslab, 2018-07-19:
OK so with Marlin 2.0.x bugfix is working, all I did is to change PS_ON_PIN   to P1_22 in Marlin/src/pins/pins_MKS_SBASE.h
On smothiee it power up but does not shutdown, I have the message on screen and I have to remove the P1_22 wire to full shutdown here is my part of config:
`#Power supply control
switch.psu.enable                            true             # turn atx on/off
switch.psu.input_on_command                  M80              #
switch.psu.input_off_command                 M81              #
switch.psu.output_pin                        1.22             # open drain, inverted
switch.psu.output_type                       digital          # on/off only
switch.psu.failsafe_set_to                   0                # so the ATX turns off on a system crash
#switch.psu.ignore_on_halt                   true             # so the ATX does not turn off on a HALT condition (like limit trigger)
                                                              # However leave commented or set to false if you want the ATX to turn off for an over heat fault condition
switch.psu.startup_state                     true`
Any idea what is missing/wrong to have full shutdown feature with smoothie?

coolt5, 2020-02-25:
A good news to everyone use MKS Gen L 1.4 with MKS TFT and MKS PWC v2
i mange to solve it .
it was the wiring from the bored MKS (PWC v2) to (MKS Gen L 1.4) the red(vvs) and black(Ground) was revers . so just swap them and it will work.

closed 2020-06-06 by makerbase-mks