makerbase-mks/MKS-TFT #113: SD Upload

vader333, 2018-02-10:
I'm using an MKS TFT 28 with MKS wifi. I can't use Wifi to upload to SD Card. Repetier host gives the error: no firmware response detected for sd card action. Pronterface is stuck at sending line 24. How can I fix this problem? My MKS wifi is configured as Access Point (AP).

makerbase-mks, 2018-02-23:
The function of SD Upload is upload to main board SD card.

iprepiprep, 2018-03-14:
Hi makerbase-mks,
could you please explain. Using repeater-host we can access the SD card inserted into MKS-TFT
- mount SD - works fine
- unmount SD - works fine
- listing SD content - works fine (see picture attached)

 Why upload does not work?

[image]

handsomedude, 2018-05-28:
Exact same issue. Also USB port don't see anything at all

odudex, 2018-07-13:
In both Pronterface and Repetier Host the SD Card upload stops at 84% and freezes. It does not delete gcode files too.

Paragaz, 2019-01-03:
Same problem here.

mcsultan, 2019-01-19:
tft35. same here, please find a solition.

furymountain, 2019-03-18:
Probably a dead thread, but for me when using Upload to SD card it starts to print over the connected port. It doesn't upload the file at all.

makerbase-mks, 2020-03-12:
To implement the function of transmitting gcode file, you need to install the MKS plugin with use MKS TFT.

closed 2020-03-12 by makerbase-mks