makerbase-mks/MKS-TFT #115: Dual Extruder Single nozzle help

wbarber69, 2018-02-18:


makerbase-mks, 2018-02-23:
1、the firmware of Marlin should to set 2 extrudes 
2、TFT_32 mks_config.txt file should to set:>cfg_sprayer_counter:2

wbarber69, 2018-02-23:
Yes I already know this.  But it still doesn’t work right.  That’s why I started this issue,  because more work needs to be done to the tft firmware.

hpkoopman, 2018-03-06:
@makerbase-mks Please fix this. Right now it's not possible to use E3D cyclops with MKS-tft;

Because there is 2 extruders, but only one heater, and one nozzle;
Now I have a useless display.

I have already set up 2 extruders in marlin, and I tried to "fool" lcd by changing temp sensor 2 (T1) pin to the same as temp sensor 1 (T0), but LCD does not recognise this.

makerbase-mks, 2018-03-07:
@wbarber69 @hpkoopman Which version of your firmware?

hpkoopman, 2018-03-07:
I'm currently on Marlin 1.1.2 and MKS 3.0; I will try with Marlin 1.1.8
this afternoon

wbarber69, 2018-03-07:
Your version of marlin has nothing to do with it.  I’m on 1.1.8-bugfix and it’s still a problem.  Hardly matters though.  This screen uses the same serial line as the usb so you can’t even use the screen and octoprint at the same time.  The guys at marlin are actually working on a 5 inch touch screen that works on i2c and spi without over burdening the serial line.  So I’ll probably just wait for that.

hpkoopman, 2018-03-07:
> safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.
> com%2Fmakerbase-mks%2FMKS-TFT%2Fissues%2F115%23issuecomment-
> 371067989&data=02%7C01%7C%7Ce4bb3212d3bd4afb672008d58408ddd1%
> 7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636560095972857493&sdata=%
> 2FAB7M2vbi3YMFKz8BeoPMgUWUokLnfXhQks5tb5_6gaJpZM4SJzK2&data=02%7C01%7C%
> 7Ce4bb3212d3bd4afb672008d58408ddd1%7C84df9e7fe9f640afb435aaaaaaaa
> aaaa%7C1%7C0%7C636560095972857493&sdata=JRFmTGEwcUaqTZ4J1R8G9FyDr4KLRK

I'm just a bit disappointed, I paid good money for this device. I'd be
better of with octoprint and touchscereen for 10 euroes more.

wbarber69, 2018-03-07:
I’m not disappointed with it it per se... it would just be a lot better if they would open up the source code or at least spend some time developing it some more.  It’s got its own processor and it can reliably send gcode as it’s supposed to, so it doesn’t fail there.  Other than the Dual extruder Single nozzle stuff, if it wasn’t such a pain to use it in conjunction with other stuff I would definitely be keeping it.  Maybe I’ll put it on one of my lesser used machines.

makerbase-mks, 2018-03-08:
@wbarber69 @hpkoopman  you can update the tft firmware to v_3.0.1

wbarber69, 2018-03-08:
Oh, really?  I didn’t know that.... it’s almost like, oh wait never mind. I’m already using the latest version! Why would that be your reply to a genuine feature request.  Now I think I will get rid of this tft screen.  Not worth one cent...

hpkoopman, 2018-03-08:
I am on 3.0.1, that's not the problem. You need to add dual extruder single
nozzle feature

makerbase-mks, 2018-03-14:
@wbarber69 @hpkoopman  You can configure the two thermal pins to be the same.

hpkoopman, 2018-03-14:
Did that. Doesn't work on TFT. Still shows two different temperatures.

wbarber69, 2018-03-14:
They just don’t get it hpkoopman

wbarber69, 2018-03-14:
Every other controller that I use that connects over serial can do this I don’t understand why it’s so hard for the mks-tft to communicate directly with marlin.  It shouldn’t be this hard.  It’s literally hard coded into my marlin firmware that any temperature setting or reading is a clone of T0.  That’s literally what the #define Single_nozzle does.  Octoprint understands this as well as pronterface.  The lcd display doesn’t have a problem it’s only the tft that can’t get its act together.

51NFUL, 2018-03-15:
I've managed to make this work by setting the temps for both extruders in the Preheat menu.  Extruder two reads that a temp is set but it is still zero (ex 0/240). Both extruders feed accordingly when extruder 1 reaches temp. This isn't a fix, but a work - around.  In no way should the people who code the board for a living consider this a fix. I'll mess with the code for the display some other time. Maybe I can make it work.

wbarber69, 2018-03-15:
It used to do this in v2 with marlin 1.1.5-6 but since they added logic for dual extruder single nozzle setups that actually works right this no longer get works in mks-tft.  I used to do what you are doing but it doesn’t work that way since I actually got my firmware setup right.  Like I said before every other device that connects over serial works like it should except for the mks-tft

51NFUL, 2018-03-18:
I've gone into the config file and edited and updated all firmware.  I printed dual extruders again tonight, only issue being over extrusion in the corners. I've printed from the touch screen, no issues.  Even added two buttons under the "more" menu to allow me to preheat to pla or abs. Now all I have to do is get a good bmp for the boot logo.

captowel, 2018-04-02:
The support for MKSTFT is thin for explaining the inner-workings.  We're on our own to find a work-around.  Here's two more options that may be helpful.

The first solution is not ideal but will work.  The TFT is looking for feedback on a temp sensor therefore, if you wire two temperature sensors to one hotend you can spoof the TFT into thinking it is heating up both sensors.

The second solution is to set both hotend temperature sensors to the same pin within marlin.  Setting both hotend temp sensor to the same pin, will return temperature readings for two hot ends woth a single sensor.  The TFT will treat your printer as if it had two hotends two hot ends with the same temp.   If you do not set the same temperature for both hot ends in your gcode, you run the risk of triggering one of the temperature safeguards within marlin.  Look for "THERMAL_RUNAWAY_PROTECTION," "MAXTEMP," and "MINTEMP" to understand how this modification may impact the safeguards built in to marlin.

Try looking in "pins.h"  

#if (TEMP_SENSOR_0==0)
 #define TEMP_0_PIN         -1
 #define HEATER_0_PIN       -1
#else
 #define HEATER_0_PIN        2    // EXTRUDER 1
 #if (TEMP_SENSOR_0==-1)
  #define TEMP_0_PIN         6    // ANALOG NUMBERING - connector *K1* on RUMBA thermocouple ADD ON is used
 #else
  #define TEMP_0_PIN         15   // ANALOG NUMBERING - default connector for thermistor *T0* on rumba board is used
 #endif
#endif

#if (TEMP_SENSOR_1==0)
 #define TEMP_1_PIN         -1
 #define HEATER_1_PIN       -1
#else
 #define HEATER_1_PIN        3    // EXTRUDER 2
 #if (TEMP_SENSOR_1==-1)
  **#define TEMP_1_PIN         5    //Change this to 6 to match HEATER_0_PIN 
 #else
  #define TEMP_1_PIN         14   // Change this to 15 to match HEATER_0_PIN**

wbarber69, 2018-04-02:
Marlin already has this workaround built in.  Literally define singlenozzle and then any temp sensor for a tool becomes T0.  Every other control method works just fine with this setup except for the tft.  A bigger pain is the absence of any real fine tuning control.  No communication with the eeprom and problematic sd card printing.  At this point I am recommending that people just don’t bother with this screen. There is no reason for it to have such a powerful platform and do next to nothing with it, and then to close off the firmware so we can’t make it worth a damn.

captowel, 2018-04-02:
> Marlin already has this workaround built in.  Literally define singlenozzle and then any temp sensor for a tool becomes T0.  Every other control method works just fine with this setup except for the [...]

I agree, marlin is already capable but as you said MKSTFT does not work with marlin's existing options.  I also agree that this screen is a poor solution to any 3D printer.  We can not send any commands to the screen through G-code so it's a waste of time.  It sells because it's pretty but for anyone who is building their own printer, the standard discount reprap LCD is a better solution since it is driven by the controller instead of being the spooler.  The real problem is MKS steals open source material then develops on top of the open source without posting it.  Sharply going against the creative commons licensing and the reprap community as a whole.

SalmosCZ, 2018-05-30:
I  dont get it. Can someone please tell me what do I have to do to see 2 extruders in MKSTFT?
I setup marlin for dual extruders, flash MKSTFT, in mks_config set 2 extruders, but I still see only 1 extruder in mkstft and can not upload filament.
MKS Gen-l 1.0 board and mks-tft 3.2 with wifi board connected.
Someting missed me?
THANKS

makerbase-mks, 2018-12-19:
You can update the firmware of the touch screen. There will be a configuration item for the dual build single nozzle in the new configuration file.
https://github.com/makerbase-mks/MKS-TFT

aaron88tk, 2019-02-22:
mks gen L V1  When I operate the  card with a 24 v power supply, there is a heat change in the extruder heaters. When I bring the temperature to 250 degrees, it drops up to 240 and it rises up to 260. makerbase-mks You can help

wbarber69, 2019-02-22:
Sound like a firmware issue not a tft issue

mcdjacesean, 2019-02-22:
Hi, I think you should use 12v instead of 24v, and I think it might be something in marlin, question what voltage heater are you using.🤔

aaron88tk, 2019-02-22:
Hi;
I have the same problem when I operate it with a 12V power supply.

aaron88tk, 2019-02-22:
Lcd12864 screen has same problem

mcdjacesean, 2019-02-22:
I recommend you to buy a new mother board

aaron88tk, 2019-02-22:
I gave you a bulk order. I have 20 cards. Most of the same problem I have tried.

mcdjacesean, 2019-02-22:
Hi have you got any pictures please

aaron88tk, 2019-02-22:
disconnected connections. when i connect again i will throw a picture

mcdjacesean, 2019-02-22:
Cool

aaron88tk, 2019-02-23:
mks gen L V1.0 card because I think I tried everything.

gysnst2, 2019-05-13:
What are we looking for in the tft firmware 3.0.3 for the 2 extruders single nozzle i see the red me states that their was an update to fix this but I’m not sure what the setting is.

wbarber69, 2019-05-13:
In the firmware it’s called “sprayers”

gysnst2, 2019-05-13:
Yes setting the to 2 sprayers still does not fix the issue with Cyclopes

wbarber69, 2019-05-13:
If you mean single nozzle then that has to be declared in marlin.  But yes the mks-tft is abysmal for 2 in to 1 setups.

gysnst2, 2019-05-13:
The newest version of the tft firmware was supposed to fix this issue.  Maybe mks can clarify to us the proper configuration settings for this issue.  Marlin is setup up properly and the config file is set to 2 sprayers.   But when you flash the tft it only shows 1 extruder?

gysnst2, 2019-05-13:
3.0.2 number 10 bug fix is this 
(10).Increase configuration items to support dual extruder single nozzle.

MKS PLEASE HELP WITH THIS SETTUP

wbarber69, 2019-05-13:
Tap the extruder icon

gysnst2, 2019-05-13:
Dude I’m not an idiot.  It’s not their

wbarber69, 2019-05-13:
It should, used to, whatever, cycle between nozzles if you tap the extruder icon.  It’s been over a year since I gave up on mine so it might have changed.

gysnst2, 2019-05-13:
I agree with you I’ve been using dual extruder for 2 years an this machine.  I want to use Cyclopes and it’s seams to be a major piano

wbarber69, 2019-05-13:
Which is why I dropped the mks tft.  It saturated the uart so using octoprint is sketchy at best when connecting.  I reverted back to the full graphics display, which has more features available anyway, and I dropped a 4”ips waveshare screen on my pi and installed it all in the same case so I get the touchscreen that’s actually functional and even has a terminal, and the instant manual controls of the lcd.  I put it up on thingiverse if you wanna look at it.
https://www.thingiverse.com/thing:2830687

gysnst2, 2019-05-13:
What lines need to be changed in marlin for the standard 2 ribbon cable lcd.  I have a few lying around I was going to use this as s last resort

wbarber69, 2019-05-13:
Depends on the screen.  Every one has a different setting to choose.  Open up your config.h and search for reprap_full that should bring you to the appropriate section.  Once you find the listing for your particular screen you can just remove the // in front of the #define for that screen.  Then just make sure you have the right libraries installed and arduino ide will automatically compile them for you.

wbarber69, 2019-05-13:
This link details the defining of such screens a little better
https://reprap.org/wiki/RepRapDiscount_Full_Graphic_Smart_Controller

makerbase-mks, 2019-05-15:
Dual Extruder Single nozzle
This configuration item already exists in tft32.
You are enabling him
There will be a temperature at the preheating interface, but two extruders can be controlled

gysnst2, 2019-05-15:
Only one extruder shows yes it it’s enabled.  That feature is way down on the Bottom of the list.  And yes 2 extruders are selected.  In repetior the temp will show on both my extruders but I only have one in the tft.