makerbase-mks/MKS-TFT #116: Disable "beep" on startup

dell9, 2018-02-20:
Hello! 

Unable to find possibility to disable "BEEP" sound on powering up. Screen model in TFT32, the latest firmware is used. 

Is it possible to remove this beep?

makerbase-mks, 2018-02-23:
You can remove the beep on the hardware.

dell9, 2018-02-23:
Ok, thank you. Will desolder beeper. But this is not good option since beeper can be useful for endusers as filament ending or power lost indicator...

closed 2018-02-23 by dell9

ozayturay, 2018-05-11:
The "booting . . ." text and the buzz is irritating and the "You can remove the beep on the hardware." is not a real solution. I guess i'll sell my tft32 and use a 128x64 graphic lcd instead. Shame on you MakerBase. :(

Evg33, 2019-01-07:
You can adjust the volume by M3 bolt in hole.