makerbase-mks/MKS-TFT #118: MKS Wifi

Thailoz, 2018-02-23:
Hi i can connect via proterface but when i send a command i get no response ive also tried with repetier host and gett a timeout error
i have a TFT32 v3.01 firm ware and an mks sbase v1,3

any ideas ?

Thailoz, 2018-02-24:
Ok done some more testing if I connect directly to the mks sbase board Wi-Fi works fine gets correct response from commands in prontaface. 
If I connect wife through the TFT I get no response to commands sent I have checked baudrates are the same

handsomedude, 2018-05-26:
Works until I disconnect Wi-Fi. Reconnect and I cannot control printer while it's printing or get any status. From Any app. Tft28 ramps 1.4 makerbase wifi

closed 2020-06-06 by makerbase-mks