makerbase-mks/MKS-TFT #119: No printing MKS TFT 32 v3.01 Marlin

tsmaa, 2018-03-04:
My equipment:
Creality v2.0 motherboard + MKS-TFT32 V3.01 + MKS Wifi + LCD12864 + 3d-Touch

Micro-sd 32gb fort fat 32 and micro-sd 64kb tested.

My firmware:
Marlin 1.1.8

All features like move, preheat, wifi access from mkscloud and pronterface are functionning OK at 115200 baud rate but a big issue discovered on print function !

When trying to print a gcode file from MKS TFT32 SD slot, after selecting & confirming the file, the screen continously increments time lapse but not printing occured after waiting a period of 20 minutes and more, no print. I can stop (cancel) the printing process at any moment. It seems like no gcode was sent to printer from mks tft32 and there is no preheat settings (from gcode) were occured. All tested gcode files were printed with success from LCD12864 SD slot and from Pronterface via USB serial baudrate 115200.

This issue (print a file) made the MKS TFT32 useless. I'm waiting for your solution after-sale ou support.

makerbase-mks, 2018-03-08:
I need your gcode file to test.

tsmaa, 2018-03-09:
gcode file sent to mailbox.
Please, check it

tsmaa, 2018-03-09:
Hi,

here is one gcode for testing MKS TFT-32

I recall: If printing from pronterface or from lcd12864, the 3dprinting is successful completed. BUt, no printing from TFT-32 v3.01

makerbase-mks, 2018-03-15:
I test it is ok with TFT32_v3.0.1 and Marlin 1.1.8.I think you need send me Marlin firmware.

makerbase-mks, 2018-03-21:
I test your gcode file is ok with the marlin 1.1.8 and tft32_v3.0.1.The mainboard is MKS_Gen_L V1.0
[https://github.com/makerbase-mks/Marlin/blob/master/Marlin-1.1.8(original).zip](url)

makerbase-mks, 2018-03-21:
I test your gcode file is ok with the marlin 1.1.8 and tft32_v3.0.1.The mainboard is MKS_Gen_L V1.0
[Marlin 1.1.8](https://github.com/makerbase-mks/Marlin/blob/master/Marlin-1.1.8(original).zip)

tsmaa, 2018-03-21:
I do not own the MKS_GEN_V1.0 mainboard.
I'm an owner of Creality V2.0 mainboard shipped with CR-10S.

May you send a debugging output version of the firmware of MKS-TFT32 which can output logs-traces messages. Then, i will send you back the logs file captured from prontoface.

Or, you make testings compatibility with Creality V2.0 mainboard.
It's up to you 

Last option is to publish the source code MKS-tft32 v301

tsmaa, 2018-03-21:
One more remark: your [Marlin 1.1.8](https://github.com/makerbase-mks/Marlin/blob/master/Marlin-1.1.8(original).zip) is not the last version available on github https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x

tsmaa, 2018-04-06:
Start printing failed with marlin 1.1.8 firmware: after confirmation of selected gcode file from microsd, no processing to print. Definitely no compatibility of MKS TFT32 3.01 and marlin 1.1.8 + Creality CR-10 v2.0 mainboard.

Checked many times with gcode files: no success.

When will you fix this bug ?

makerbase-mks, 2018-04-08:
You can replace it with a small SD card.

tsmaa, 2018-04-08:
NO SUCCESS: already checked with many capacity of sd card: 8 GB, 16gb, 32GB of brand Sandisk, Kingston, Transcend. If i used these cards from a LCD12864, the printing function is completed successfully but NOT from MKS TFT32 v3.01 !!!

MKS TFT32 v3.01 is definitely buggy when i confirm printing any gcode file !

makerbase-mks, 2018-04-08:
I test your gcode file is ok with the Marlin_1.1.8_bugfix_1.1.x and tft32_v3.0.1.The mainboard is Creality V_2.1
[Marlin_1.1.8_bugfix_1.1.x](https://github.com/makerbase-mks/Marlin/blob/master/Marlin-bugfix-1.1.x(20180120).rar)

BlackWolf57, 2018-04-13:
Same for me.
The only way is to set TFT at 250000 baud and it work.
But if you are on Linux like me 250000 baud seems don't work on USB serial and you should make a choice.

Not tested with baudrate lower than 115200

tsmaa, 2018-04-16:
Thank you BlackWolf57 for your suggestion.

I confirm that at 250000 baudrate, the 3dprint from TFT32 SD support succeded. Eureka !
But there is a drawback: Pause/resume printing loops occured  randomly (message: checksum error and resend line number xxxx) if Printrun (Prontoface) is connected on port com3 at same time.

So, there is still a bug with MKS TFT32 printing process with low baudrates. Normally, lower baudrates must be more accurate and stable but not with this firmware. Serial conflict ?

lifeandfun, 2018-05-17:
Same for me,too

I have tested my MKS TFT24 v1.0 at 250000 baudrate already but nothing happen , my 3d printer still not receive gcode form SD card.

Because MKS TFT24 English firmware was deleted, so i try to replace it with MKS TFT28/32 English firmware. Manual control is ok, but can not print with SD card. Maybe this is the reason.

Please share me MKS TFT24 English firmware. Thank you so much.