makerbase-mks/MKS-TFT #120: TFT32_L v3.0 not booting - just beeps every few seconds. Bootloader required.

voodorush, 2018-03-05:


voodorush, 2018-03-07:
Is anyone reading this? The mainboard states version 3.0 at the element size. Ihave ordered alredy STlink v2 so everything i need is a binary file for upload. No matter if its only the bootloader or entire firmware. There is no "booting" green sign so obviously its the bootloader.

voodorush, 2018-03-08:
No. Thats the point. The problem is in the bootloader part of the firmware. But the firmware here doesnt contain bootloader section. It is designed to be uploaded by the bootloader. I need the firmware including bootloader or the bootloader alone. The factory uploaded fimware (no matter what version) will do trick - i believe its uploaded via the JTAG interface jus like i'm going to do.
It's good idea to have standalone bootloader or firmware with bootloader for direct upload via the JTAG pins here on Github. It can help other people too.

voodorush, 2018-03-13:
Are you going to upload or send to my email the bootloader or i should return the display to  the 3d printer manufacturer and he will send it to you for repair? I was trying to skip this part since i can fix it and save time for all of us - the procedure will take almost 2 month knowing how fast works the china airpost.

makerbase-mks, 2018-03-21:
You need  contact 3d printer manufacturer  replace the display.

closed 2018-04-26 by makerbase-mks

labalast, 2019-05-08:
I have the same problem, but I cannot ask the manufacturer to replace the display, because I bought it
separately. Why can´t you just send a copy of the bootloader?