makerbase-mks/MKS-TFT #130: Please consider releasing the source code

superdbop, 2018-04-17:
Hello, I just purchased a TFT32 and I'm very excited to use it. However after reading through the issues on your Github page, my excitement is beginning to weaken and I'm considering returning it..

While browsing the issues list here on Github, I can't help but notice how many features are not being implemented due to the developers at makerbase-mks finding them difficult to code or sometimes not understanding the bug or feature request at all. Issues that could easily be solved by allowing other people from around the world help out and make your product even better.

I'm asking, would you please consider releasing the source code for your TFT LCD controllers? By releasing the code, other people from around the world will be able to help you fix bugs, implement new features, and overall help your products reach their full potential (which will ultimately result in more sales.)

As I recently found out, when you turn on a MKS TFT screen, you're greeted with the words "RepRap" in big bold letters. Please let those in the RepRap community help make your products even better.

Thanks

makerbase-mks, 2018-04-26:
We will consider it.

closed 2018-04-26 by makerbase-mks

superdbop, 2018-04-26:
That sounds great, thank you. I really hope it happens. The MKS-TFT has so much potential.

openbuilds-engineer, 2018-11-07:
Hi @makerbase-mks is the source available yet?  As a developer I can and want to contribute and help make it better for you.  If you send me source, I will send back fixes and new features to you?

wildblue, 2020-01-28:
I would like to see the sourcecode, too. BigtreeTech did so for their display and i am considering to switch my hardware because of this fact.