makerbase-mks/MKS-TFT #131: problems with the products

eduardoluque, 2018-04-24:
Hello good afternoon,

I am writing to you because I have several failures or problems with the products that I have acquired from your signature on aliexpress and they do not know how to respond.

I have bought several products, I specify:

   - Motherboard MKS GEN L V1.0
   - Display Reprap LCD 12864
   - LCD Touch Screen MKS TFT 2.4
   - Wifi module MKS TFT hlkwifi V1.1

Well, I'll tell you about the problems I have:

- First of all the Reprap LCD Display 12864 I am not able to put it into operation with the base card MKS Gen L V1.0 does not work, however with an Arduino Mega 2560 board it works perfectly. For both I use the Marlin firmware.

- Secondly, when I print directly from the SD card of the LCD Touch Screen MKS TFT 2.4, the printer starts working but after a few seconds it hangs and stops working, however from the PC with Pontiforce and Simplify works perfectly .

- And finally the Wifi Module MKS TFT hlkwifi V1.1 does not work with the LCD Touch Screen MKS TFT 2.4 I have installed the firmware 3.0.1 whenever it connects brand status of Exception and I am not able to connect with it. I have also installed the firmware 2.4 and with this other I can not connect, always the status is disconnected.

I attach the Marlin file that I have on the Motherboard MKS GEN L V1.0

Please help me to correct these mistakes.

Greetings and thanks.

makerbase-mks, 2018-04-26:
1、Which version of your hardware of TFT24?
you can give me your marlin firmware to test.

eduardoluque, 2018-04-27:
Hello good day,

First thank you for your prompt response.
I attached the Marlin file that I have installed.
The hardware version is MKS TFT24 V1.0 and the firmware I am using is V3.0.1 I send the config file that I am using. You could also tell me why the MKS TFT module HLKWIFI V1.1 does not work for me

Thank you very much for everything, I await your response
[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/1954644/mks_config.txt)
[SkynetV2.2_Edu_Ok_01_MKS_BASE_PID_BED___FULL_GRAPHICS_2_EXTRUDE.zip](https://github.com/makerbase-mks/MKS-TFT/files/1954645/SkynetV2.2_Edu_Ok_01_MKS_BASE_PID_BED___FULL_GRAPHICS_2_EXTRUDE.zip)

Thirkell, 2018-05-04:
In regards to your problems from another disappointed persons view:
1.      Some of the LCD 12864 screens have the connectors the wrong way around. There are some forums and videos on you tube explaining this and showing the fix. I had to do it to get my GLCD working with the SBase board

2.  The print pausing from SD print on the TFT screen is a very common and unresolved issue. Nothing useful on fixing it or indication that they are trying from Makerbase

3.  Cant offer advice here. I have not even attempted trying the wifi thing as the SD print makes the entire screen almost useless from my point of view

eduardoluque, 2018-05-07:
Good morning Thirkell,

Thank you very much for your response and your help.

1.- Indeed, the connectors are reversed. Incredible, in my Arduino 2506 work just the opposite, I have solved by putting new connectors.

2.- It's a pity that this screen does not work. It looks very good and could be very useful.

3.- And finally, as you say, the Wi-Fi does not help until the pause is solved or the SD hang up.

Thanks again

eduardoluque, 2018-05-08:
Good afternoon,

With the latest firmware updated on the V3.0.2 page the printer does not crash and works perfectly with Marlin. What is still not working is the Wifi module.

Greetings

t0mac0, 2019-01-22:
I have a Prusa I3 hephestos, I bougth MKS Gen L V1.0 and TFT32_L V4.0, I download Marlin and configure with Prusa parameters, But the Z Min Stop it doesn't work, the X and Y axis homming works perfectly but the Homming of Z axis is not working, I tried config the inverting flag in config configure.h firmware, but nothing, I'm thinking that is the input pin is dead.