makerbase-mks/MKS-TFT #133: MKS TFT35 - Size for the booting logo and buttons

Timppa71, 2018-05-04:
Hi.

I did not found any information what is the right size of the booting logo for MKS TFT35? Also, what is right size for button pictures? Can you share this information, please?
Is there datasheet for the product?

makerbase-mks, 2018-05-14:
the button pictures size is 117*140 ,and logo is 480*320.

Timppa71, 2018-05-14:
Hi.

Thanks for the info.

However, this is strange because I can make a custom buttons (with size 117x140px) for any other buttons except "More" buttons (bmp_morefunc1 to bmp_morefunc7)? They just don't show correctly. Only button picture which is shown correctly is "zoffset" button. And same problem with "Custom" button pictures.

[image]

makerbase-mks, 2018-05-15:
bmp_custom1.bin to bmp_custom6.bin

Timppa71, 2018-05-15:
Yes, indeed. My bad.

Great support, thank you so much!

closed 2018-05-15 by Timppa71