makerbase-mks/MKS-TFT #136: Extent lenght of filament load/ retract?

Timppa71, 2018-05-05:
Can you improve your FW-code like it would be possible to feed more filament under "Settings - Filament - Load / Unload"?
Long retraction is prohibited in my Repetier-software and maximum feeded lenght of filament is 150-160mm.
However, I can make my own "more-item" button and avoid this limitation by feeding filament several times of needed lenght... For example 150mm + 150mm + 150mm. But like this I will two "More" buttons (and More button no. 7 is not even work :), so this is not a very good option, I think) 

Can you modify you FW to get this same kind of behaviour? For example -> cfg_filament_load_lenght:150+150+150 will feed filament total of 450mm...

makerbase-mks, 2018-05-14:
Can't it be achieved by clicking three times?

Timppa71, 2018-05-14:
Yes, but I solve this by modifying FW-code of my arduino. Case solved.

closed 2018-05-14 by Timppa71