makerbase-mks/MKS-TFT #138: MKS tft and Octoprint Control over USB

merlin2105, 2018-05-05:
my MKS tft is now running on a Ramps 1.4, if I have connected the tft the control via USB (Octoprint) does not work anymore ... is that the same USB port ..? can one adjust this in Marlin?

makerbase-mks, 2018-05-14:
USB and TFT serial ports are shared and cannot be controlled at the same time.

merlin2105, 2018-05-27:
works this with an MKS SBase Board , the tft an Octoprint over USB at the same time..?

sb43201, 2018-06-11:
Could you add a function to disable TFT serial communication to let print from USB port?

pyr0ball, 2018-07-25:
Would it be possible to connect an OctoPi to the MKS TFT WiFi UART pins to control it that way?

philoupat, 2018-12-12:
Pins above aux2 indicate RX2 TX2 WIFI

jimmyeao, 2019-01-30:
So do I read this right? If we have a MKS TFT board plugged in, we can no longer use Octoprint to print via usb? This would explain all the crc errors i am seeing

Andrei-Pozolotin, 2019-03-06:
See #285

didtoknan, 2019-03-24:
The TFT32 is connected to the AUX 1 MKS Gen L port. According to the schematic, 2 pins are 5V, 2 are Ground, 1 is D1 and the last one is D2. What about wiring a dpdt switch to disconnect for example D1 & D2 ? We could flip that switch before starting a print from octoprint. Once is done and we want to use the TFT32, we will just have to switch the flip the other side. In fact we may only need a spdt (or spst) switch to disconnect the TX pin. I thought the MKS Gen L was based on a Arduino Mega but it looks I'm wrong because serial pins on Mega are D0 & D1, not D1 & D2.

didtoknan, 2019-03-24:
According to the MKS Gen L datasheet TX is D2, RX is D1.

Stady234, 2019-04-07:
I maybe out on a limb by myself... but I have been printing via octoprint on an MKS Gen L with the TFT3.2 connected without issues... 
Are you sure you have everything connected the right way?

Muskateer10, 2019-05-23:
My OctoPrint works. The problem is every time I want to make a change to Marlin I have to disconnect the TFT32 display. It is because there is a conflict with D1 and D2 on Aux1 (which is also the USB port).

Stady234, 2019-05-28:
@Andrei-Pozolotin I see that you keep posting for an alternative fw solution... does the CNC version you have referenced in the thread work for us normal printer users?

@Muskateer10 I have the same issue... i use the remote firmware push with octoprint, i have a note on my screen to disconnect the TFT before the push, then back to normal operation...