makerbase-mks/MKS-TFT #144: TFT24 fw 302 Russian language

LyaminSS, 2018-05-14:
Russian language
the letter "Я" is displayed incorrectly
How to fix?

[image]
[image]
[image]

LyaminSS, 2018-05-14:
version 301 has this problem too

LyaminSS, 2018-05-14:
that's how it will be correct
[image]

makerbase-mks, 2018-05-15:
we'll fix the bug recently.

LyaminSS, 2018-05-20:
I noticed that all files on the server were deleted and new ones were added.
I flashed display
The problem is not solved.

LyaminSS, 2018-09-30:
[image]

LyaminSS, 2018-10-23:
3.0.3 not fixed!
Give a solution!

makerbase-mks, 2018-10-24:
We will update and resolve this issue soon.

makerbase-mks, 2018-11-14:
We have modified it.

LyaminSS, 2018-11-14:
[image]
Thanks!

closed 2018-11-14 by LyaminSS