makerbase-mks/MKS-TFT #145: MKS SBase 1.3 + MKS TFT32 USB and SD printing just stops in the middle of a print

FloxRoxX, 2018-05-16:
As mentioned in the topic:
Hardware: 
-MKS SBASE 1.3 with latest Smoothieware firmware
-MKS TFT32 with firmware 3.0.1
-Original cable used that came with the board
Slicer:
Simplify 3D

Problem: The print just stops in the middle of printing the first layers and will not continue from then on.

What can we do about this issue? I already consulted other issue threads but all i found was for marlin boards and not for smoothieware...

makerbase-mks, 2018-05-25:
Have you tried to print your GCODE file with the host computer?

makerbase-mks, 2018-05-25:
There may be static interference problems, you can try to use the serial cable and shielding.

Mosfett1975, 2018-07-25:
Have the some mulfunction. I've been made very short cable betwin SBASE and TFT, set 9600 baund rate - doesn't help

makerbase-mks, 2018-07-26:
@Mosfett1975 what is version of your TFT software?

Mosfett1975, 2018-07-26:
it's TFT32 v3.0.2

closed 2020-06-06 by makerbase-mks