makerbase-mks/MKS-TFT #147: TFT32 Blank white screen

FBReality, 2018-05-22:
Hello
My screen stay full white, when I reset or restart, I have a white screen, a black , a beep and return to white, nothing more
I tried to update firmware but nothing change
One time when I pressed on screen, my printer start to move but I don't know what I do to have this.
Seems screen work but show nothing.
If you have any idea
Thanks in advance

makerbase-mks, 2018-05-25:
What is the version of software and hardware.

FBReality, 2018-05-25:
Hardware 3.0
Software, I tried 3.0.1 and 3.0.2, same result

makerbase-mks, 2018-05-29:
The screen may be damaged, you can contact our sales replacement TFT.

mageta90, 2018-05-29:
Hello i have the same issues.

screen built in in a jgaurora printer. worked fine for 3 months, then out of the blue the screen is all white.
tried to upload the original firmware, then the 3.0.1 but no change.

the same behavior of FBReality's one

hardware is 3.0

The firmware is actually updating, because the files in the SD get names changed .

mageta90, 2018-06-11:
no idea of what's appening here?

Vorchak, 2018-08-02:
Is there an update on this issue? I have the exact same symptoms. The screen is all white, however, I noticed that the icons are still there just not visible. And they even respond to touch.

FBReality, 2018-08-02:
No news, finally I bought a new one that works perfectly, and from a shop where I can get a real warranty

closed 2018-08-02 by FBReality