makerbase-mks/MKS-TFT #149: Mks Tft28 makerbase Wi-Fi print stopping at start

handsomedude, 2018-05-27:
Got a tft28 ver 2 3.0.2 firmware with makerbase Wi-Fi and ramps 1.4 Marlin 1.1.8
Port baud rate 250000

Prints fine from sd card.  

When I send a print job from a PC or phone, the job starts and then stops a minute later and just sits there and does nothing.

Also as card upload, won't upload.

handsomedude, 2018-05-27:
I've installed Wi-Fi firmware from 3.0.1 and now it prints from Wi-Fi fine.

Only issue is, PC repetier don't show heater heating. Reaches temp then updates.

If I use Android app and reconnect during a sdcard print, I cannot see anystatus of the progress only temp.

sharkovios, 2018-05-28:
The same problem with 3.0.0 & 3.0.1.
When printing from USB print stop at start.
When printing the same gcode file from sdcard, everything's fine

makerbase-mks, 2018-05-29:
@handsomedude Can you use 3.0.1 firmware normally?
@sharkovios you can try it replacement the usb2.0

handsomedude, 2018-05-29:
I've tried firmware 2.0.1  3.0.1 3.0.2

All do same thing.  I have two Wi-Fi modules. One is firmware updated and other is as it came.

Old Wi-Fi firmware, PC connects and that's it.
New Wi-Fi firmware, PC connects and sends commands but send a print and sometimes it will just home axis, other times it starts a print then just stops about a min later.

Think issue is the Wi-Fi firmware.

Are there any other firmware updates for Wi-Fi?