makerbase-mks/MKS-TFT #151: HLK-WIFI Connection

nip2k2, 2018-05-31:
Hi! 
I have HLK-WIFI connect to a TFT3.2 and when I try to connect to a router though the parameters on the mks_config.txt file it says Exception and wont connect (this happend on some routers, in others work fine).
But if I put as an AP, I can connect to the printer and then in the web interface modify the parameters to a station and connect to the router that have the Exception problem. Anyway when I reboot the printer, the parameters come back to the ones on the mks_config.txt previously loaded.
So, in my opinion, It seems that we need an extra parameter like encrypt type to solve this in the mks_config.txt that is loaded. Also would be good to have an DNS parameter too because by dafault is 192.168.3.1.
Maybe that parameters already exists, I tried to find them with no success. Sorry my english!
Thanks in advance!

nip2k2, 2018-06-09:
Any comment about this?

makerbase-mks, 2018-06-11:
1, AP mode does not need to modify the IP address, the default is 192.168.4.1

closed 2018-07-16 by makerbase-mks