makerbase-mks/MKS-TFT #152: Firmware 3.02 USB stick does not work

aurion55, 2018-06-01:
Can you please explain how to get the USB stick to work, I can read via SD card, but NOT a USB stick, I had used every firmware version and none of them work.

aurion55, 2018-06-01:
[image]

aurion55, 2018-06-01:
Well it was suppose to been fixed at version 1.2.0, I'm using V3.02

handsomedude, 2018-06-01:
Exact same issue. USB never worked at all.

Tft28 vet 2.0 

Tried all firmware available

handsomedude, 2018-06-01:
Ok. Had a brain fart lol

Just received a tft3.5 ver 1.0 .  Did same thing.

Goto settings, file, and select USB lol.

That's it.  Both work fine.  Silly both can't be mounted same time!! 

I can finally use USB!!!

aurion55, 2018-06-01:
I've been into setting and selected USB it does not work, if I insert a USB stick.
What size USB stick you're using.

handsomedude, 2018-06-01:
I've tried a 2gig sdcard in a USB adapter.  I've also tried my 32gig USB 3.0 stick.  Both work fine.

One is fat file sysyem and other is fat32

aurion55, 2018-06-02:
Item 9 still not fixed!!!

 V3.0.2 firmware modification(2018/05/05)

(1).Add the "M-off" button to the "Home" interface

(2).We changed default temperature 180 to 0 in the "Preheat" interface

(3).Add the function of display mainboard eeror

(4).in the fan inerface,press the "+" or "-",the fan work

(5).fixed the bug of  click "more" restart  

(6).The board will stop operating more than 5 times when using the M110 re-instruction command

(7).Increase the configuration item to shield power failure detection function

(8).Modify the bug that serial port interrupt sending will be interrupted

(9).Modify the bug of read failure in U disk or SD card printing

(10).Increase configuration items to support dual extruder single nozzle.

(11).Add a button in the WIFI interface to control wifi module  function open or close

(12).Fixed bug that the simple version and the retro version manual leveling wasn't go home

(13).Wifi firmware was fixed bug that function incorrect when password less than 8 digits

aurion55, 2018-06-02:
Try a different USB stick and it worked.

closed 2018-06-02 by aurion55