makerbase-mks/MKS-TFT #153: Flip screen upside down

Henry2501, 2018-06-01:
Good Day. I have a MKS Robin board with TFT 3.2 screen. The way the machine is built i need to install the screen upside down in the machine due to the length of the flat cable only being 300mm. 

Is there a way to turn the screen 180Degrees in the config file or firmware. 

Urgently looking for help, i need this done on 6 machines.

Timppa71, 2018-06-01:
Yes, you can do it inside "mks_config.txt" file. Scroll down and you can find the parameter almost at the end of file...

Henry2501, 2018-06-01:
Hello Timppa71

Thank you for the reply. I posted this in the wrong section. I have a MKS
Robin Board that comes with a tft screen. It does not have that option in
the Robin Config files. Tried even the latest config 2.0.7.
I also Added the code
>cfg_OVERTURN_180:1 #The screen is flipped 180°(0: no flip, 1: flip)
to the Robin config file but did not work.

makerbase-mks, 2018-06-11:
Do you use MKS TFT32_L or MKS Robin TFT?

makerbase-mks, 2018-06-11:
The MKS Robin TFT have not function of screen overturn 180 .

closed 2018-07-16 by makerbase-mks