makerbase-mks/MKS-TFT #154: MKSTFT28 move axis

andreyk0, 2018-06-03:
Hi! I have this TFT connected to Marlin firmware.
When I try to move X or Y axis manually from the UI the movement only happens in one direction, so, e.g.  trying to move X left or right only moves it right with each press, both buttons.

Strange issue, not sure how to fix. When connected to Marlin over USB port via terminal and moving with G01 commands everything works fine.

Please advice.

makerbase-mks, 2018-06-11:
The screen just send command:
G91
G1 X10 Fxxx
G90
or 
G91
G1 X-10 Fxxx
G90

andreyk0, 2018-06-19:
Yep, you are right, firmware is behaving correctly, turns out my Marlin code is only working properly with absolute coordinates. Switching to relative coordinates I observe the same behavior when issuing commands manually via serial console. Thanks for your help, I'll debug Marlin now.

closed 2018-06-19 by andreyk0

andreyk0, 2018-06-20:
Ok, just in case someone else comes across this, bug is related to the application of the home offsets via M206. The bugfix-1.1.x branch of Marlin seems to behave correctly compared to 1.1.8 release version.