makerbase-mks/MKS-TFT #156: Z Offset Calibration using TFT 24

CRojasV, 2018-06-08:
Hi, we want to know how we can do to calibrate the Z-offset using the tft screen. This is an important task that simple prusa printers can do using a cheap rep  rap Full Graphic Smart Controller. 

We currently use the following gcode to do it using a serial prompt:
 M851 Z-1.1 #Set the Z offset to -1.1 mm
M500 #To save changes

But We need to do it using the tft, can we do it ? Can you help us with this issue?

makerbase-mks, 2018-06-11:
You can modify the configuration file:
>moreitem_button1_cmd:M851 Z-1.1;M500;

CRojasV, 2018-06-13:
Sorry, but that solution doesn´t work properly because using that line you cant´t modify the Z offset value from the tft interface, and you need to modify periodicadly the tft configuration file to ajust the Z offset. Can we add a custom menu to do this? It posible to have a prompt interface in the tft?

makerbase-mks, 2018-07-09:
TFT35 can achieve this function.

closed 2018-07-16 by makerbase-mks

1lug1998, 2021-11-18:
I have this exact same need. Can TFT32 do this?