makerbase-mks/MKS-TFT #157: TFT35: Error:Checksum required when switching back to ASCII protocol

Udo-Nbg, 2018-06-08:
MKS TFT35 + MKS-Gen 1.4 + Repetier 1.0.1

Repetier-Firmware configuration tool for version 1.0.2
TFT35_V1.0.1_English

While Testing i got 
**Error:Checksum required when switching back to ASCII protocol.**
in Repetier Host and also on the TFT

Alwas i send a command via Touchsreen (example): "Settings/Fan/100%" OR "Settings/Fan/0%" 

@115200 ANSI 
@250000

GEN & TFT baudrate alsways change on both devices.

Force checksums once a checksum is received ist set

makerbase-mks, 2018-06-11:
Do you selected 2 for this configuration:
>cfg_firmware_type:2

Udo-Nbg, 2018-06-11:
## Of course,

i tried also older Marlin Firmware Version 0.92 or so.... but nothing better happen

also i reboot more than one time the controllers (board and tft).

maybe i have an other idea but i'm actually not at home so i can test in a vew days.

**Where can i find the complete user manual for the tft35**

and the software **mks-host** is onli with chinese letters, is ther a other **english version** available?

Udo-Nbg, 2018-06-14:
Problem probably solved,
Errors appear when concurrent access via the HOST software AND commands from TFT Dispay take place.

Unfortunately, I can not find a data sheet or manual for the TFT35 yet.

makerbase-mks, 2018-07-16:
https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20TFT3.5%20Datasheet.pdf

closed 2018-07-16 by makerbase-mks