makerbase-mks/MKS-TFT #158: Tft 3.5 baud rate issues

handsomedude, 2018-06-11:
I set my arduino to 250000 and the display the same.
Was all working fine.

I needed to make adjustments in Marlin so I flashed arduino using uart back to 115200.

Now when I flashed Marlin back with port see at 250000, and the screen the same, the LCD will not command the display.

Marlin 115200 + LCD 115200 commands but print just stops randomly and don't restart.

Marlin 250000 + LCD 115200 all works fine

Marlin 250000 + LCD 250000 nothing works.

Tried reflashimg multiple times.  Still same results.

Now set at Marlin 250000 and LCD 115200

Seems like the LCD has to be set half the Marlin speed for it to function fully

Hope this helps someone

makerbase-mks, 2018-06-11:
What is your tft3.5 firmware version?

handsomedude, 2018-06-11:
Isn't there only one version firmware available??

makerbase-mks, 2018-06-11:
The latest firmware is V_1.0.1.

handsomedude, 2018-06-11:
That's the o ly firmware available for download for this model.

closed 2018-07-16 by makerbase-mks