makerbase-mks/MKS-TFT #159: TFT 35 All Icons and Images Sizes to Custom UI

SpecialFx78, 2018-06-15:
Hello, 

Recap of all different size of image to create personal UI for TFT35

LOGO after Boot : 480x320 bmp_logo
Main icons size : 117x140
Other "bmp_pageUp ; bmp_pageDown ; bmp_return" : Unknown
Other "bmp_pause ; bmp_stop ; bmp_menu" : Unknown
Other small icons uses during printing : Unknown

Please if you have the possibility to communicate the missing size ? thank's

Regards

Operrr, 2018-06-18:
Hi. bmp_preview 200x200 pix

SpecialFx78, 2018-06-21:
Hello,

Would it be possible to answer this issue? please.

Can you tell us the sizes of the different icons of the image capture below.

[image]

N1 "bmp_pageUp ; bmp_pageDown ; bmp_return" 
N2 "bmp_pause ; bmp_stop ; bmp_menu" 
N3  small icons uses during printing 

Thank you for your return, I am blocked to finish my interface

Regards

Operrr, 2018-07-12:
See here https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20TFT3.5%20Datasheet.pdf

SpecialFx78, 2018-07-13:
Thank you for the English translation of the manual, but I just read it and I still can not find the information requested.
It seems to me however clear what is asked.
I even put screenshots.
What is complicated in the application?
I am just asking for the sizes of the images in pixels, the images whose sizes are unknown are indicated on the catches.

Thank you for taking the time to get closer to the developers to obtain the information or to indicate to us the sizes in pixels or to provide us with the AI ​​sources of all the images used in the interface of the TFT 35.

Very cordially

Operrr, 2018-07-14:
Name | Resolution
-- | --
bmp_logo | 480x320
bmp_preview | 200x200
bmp_operate | 150х80
bmp_pause | 150x80
bmp_printing_back | 150x80
bmp_resume | 150x80
bmp_stop | 150x80
  |  
Main icons | 117x140
  |  
bmp_back | 117х92
bmp_pageDown | 117х92
bmp_pageUp | 117х92
  |  
bmp_function2 | 100х100
bmp_menu | 100х100
bmp_speed_extru | 100x100
  |  
bmp_zpos | 45x45
bmp_time_state | 45x45
bmp_bed_state | 45х45
bmp_ext1_state | 45х45
bmp_ext2_state | 45х45
bmp_fan_state | 45х45
bmp_speed_state | 45x45

SpecialFx78, 2018-07-14:
Thank you very much

burakerenn, 2018-11-07:
How can I see models in tft. Like in the picture. (number 4)

sandog78, 2019-02-07:
Is there a way to get a custom UI in English from someone?

venkatkamesh, 2019-06-17:
[image]
[image]

I Have tried changing theme and icons for my tft35..
everything changed except the background color of the text like in images
is that a bug or any solution for it ?

your help is really appreciated 
Thank you @Operrr .....

MarshMalloW0007, 2019-12-20:
Hello,
Read my ([issue](https://github.com/makerbase-mks/MKS-TFT/issues/359)) to solve your problem.
Marsh

closed 2020-06-06 by makerbase-mks