makerbase-mks/MKS-TFT #161: MKS TFT 3.5 Datasheet

Operrr, 2018-06-18:
Hello! Why did not you put the datasheet on this version? I'm very interested in the size of the icons for creating a custom interface.

Operrr, 2018-06-25:
Do the implementers generally visit this place? there are not many answers, all died?

schmitzCatz, 2018-07-05:
they deleted the datasheets, unfortunately the TFT 3.5 never got uploaded. Maybe the TFT 28 can help you out: https://github.com/makerbase-mks/MKS-TFT/tree/600550aea78e47ddc3311697ec65cffbe52e51bf/MKS-TFT2.8-3.2

makerbase-mks, 2018-07-11:
Due to our updated the datasheet recently, no instructions have been uploaded.
https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20TFT3.5%20Datasheet.pdf

Operrr, 2018-07-12:
Thank you! Where can I download the plug-in for the SURA slice?

>Среда, 11 июля 2018, 18:53 +08:00 от makerbase <notifications@github.com>:
>
>Из-за недавно обновленного технического описания никаких инструкций не было загружено. 
>https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20TFT3.5%20Datasheet.pdf
>- 
>Вы получаете это, потому что вы создали тему. 
>Ответьте на это письмо прямо,  посмотрите на GitHub или  отключите тему .

schmitzCatz, 2018-07-13:
any chance on releasing the source code for the modules?

makerbase-mks, 2018-07-16:
https://github.com/makerbase-mks/Software/tree/master/MKS%20Plugin

Operrr, 2018-07-16:
Hello! I apologize for persistence. The plug-in 3.X works in versions prior to 3.2.1. In newer versions, the plugin can not save the file. This is not critical, but I would like to use the latest version of the slicer.

makerbase-mks, 2018-07-30:
@Operrr  We have updated the plugin:
https://github.com/makerbase-mks/Software/tree/master/MKS%20Plugin

ivanhauptman, 2018-08-29:
Could you please tell me the difference between the MKS TFT 3.2 and 3.5?
And just the size?

Operrr, 2018-08-31:
not only, 3.5 "can display a preview of the printed model

kpdelucia, 2019-02-02:
Any instructions on how to install this plug in?  The readme file was blank in the repository

closed 2020-06-06 by makerbase-mks