makerbase-mks/MKS-TFT #163: MKS TFT7.0 - Rotate Screen

gorbi87, 2018-06-22:
Hi,
is it possible to Rotate the Screen by 90° to install the TFT vertical?

makerbase-mks, 2018-07-11:
You need to custom designed for it .

makerbase-mks, 2018-07-16:
King@makerbase.com.cn

gorbi87, 2018-07-20:
How can i customize it? How can i change the Button Positions for example? The manual shows only how to change background, color etc.

Andrei-Pozolotin, 2019-03-06:
See #285

closed 2020-06-06 by makerbase-mks

mshavandi, 2020-10-10:
@Andrei-Pozolotin I checked that link, but it seems that it's just an introduce to the firmware and again no guide to change it.
I tried building the firmware with no changes in the code, the generated `.bin` file made my lcd stuck at `100% uploading`. I searched further and found out that it's got to be decoded with a simple tool dedicated for the `mks robin nano`. I used that and all the screen got mixed up, images got corrupted and the dimension of the graphics reduced to the size of `mks robin nano`. So I found out that it's ultimately useless for `mks tft 7.0` which is my lcd.
In this interval I have talked to many `mks tft` lcd buyers and understood that they have the same problem. Please some one help us how to make changes in the firmware and the most important, **provide a .bin decoder for mkstft70**. Please!! @makerbase-mks