makerbase-mks/MKS-TFT #167: tevo little monster TFT Problem all sideways

rothlive, 2018-07-03:
Since I installed the update to version 3.0 from version 1.2 everything is wrong on the screen. Unless I'm using a version larger than 3.0.

But i need to use it with tevo Firmware and it gave no Version over 3.0

[image]

rothlive, 2018-07-04:
I have testet a little further.
Tevo Version based on Version 1.2 from 2016.

All versions below 3.0 have the error.
Also the versions of https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT2.8-3.2/History%20version

Hardware defect?
I dont know.
For it to work again, a new firmware would have to be created based on TFT28_32_v3.0.2 Release file.zip.

Or another hardware unfortunately.
I also do not understand why that is.
https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT2.8-3.2

rothlive, 2018-07-07:
Tevo send me a new Firmware and then it works fine !

closed 2018-07-07 by rothlive

Prcrstndr, 2019-02-01:
> Tevo send me a new Firmware and then it works fine !

Hello, I'm having problems with the Tevo Little Monster, can you share your new firmware please?

rothlive, 2019-02-01:
> 
> 
> > Tevo send me a new Firmware and then it works fine !
> 
> Hello, I'm having problems with the Tevo Little Monster, can you share your new firmware please?

yes
How can I get your e-mail address at Github?

Prcrstndr, 2019-02-01:
> > > Tevo send me a new Firmware and then it works fine !
> > 
> > 
> > Hello, I'm having problems with the Tevo Little Monster, can you share your new firmware please?
> 
> yes
> How can I get your e-mail address at Github?

dgonzalez1013@cova.edu.mx
Thank you :)

rothlive, 2019-02-01:
You have a Mail

ducmphung, 2019-03-25:
Hey guys I am having the same issue. Would you mind sending a working version to me also please? duc@ducphung.com thanks in advance.

rothlive, 2019-03-26:
Ok

mzaczynski, 2019-05-03:
Hello guys, 
my TLM also have the same issue, could you share with me your working firmware please?
My email address at github is: mzaczynski@gmail.com

mzaczynski, 2019-05-03:
Thanks in advance

rothlive, 2019-05-03:
You have a Mail with the firmware.

> 
> 
> Hello guys,
> my TLM also have the same issue, could you share with me your working firmware please?
> My email address at github is: [mzaczynski@gmail.com](mailto:mzaczynski@gmail.com)

myeoman101, 2019-06-01:
Hi I'm also having same issue, could you email me the firmware at myeoman@hotmail.com. Thanks

bestia-io, 2019-09-16:
hi, same problem on my tlm. Care to share? jtravis.russett@gmail.com thanks!

PterodacylDanceParty, 2020-08-15:
> You have a Mail with the firmware.
> 
> > Hello guys,
> > my TLM also have the same issue, could you share with me your working firmware please?
> > My email address at github is: [mzaczynski@gmail.com](mailto:mzaczynski@gmail.com)

Hello, could you please send the new firmware to robert.m.busic@gmail.com? I ran into this same issue

rothlive, 2020-08-16:
Hello

 

Yes.

 

With best regards, Markus Roth from Germany.

 

Hello, could you please send the new firmware to robert.m.busic@gmail.com <mailto:robert.m.busic@gmail.com> ? I ran into this same issue

docman11, 2025-04-15:
Hi! Can u help me please with working firmware? Same problem... 
Thanks in advance.
My email: docman545@gmail.com

PterodacylDanceParty, 2025-04-15:
> Hi! Can u help me please with working firmware? Same problem... Thanks in advance. My email: [docman545@gmail.com](mailto:docman545@gmail.com)

It never worked. I gave up and got rid of the machine. I think if I still had the machine I’d convert it to Klipper and just get some cheaper MKS board on Amazon.

rothlive, 2025-04-19:
> Hi! Can u help me please with working firmware? Same problem... Thanks in advance. My email: [docman545@gmail.com](mailto:docman545@gmail.com)

I have sent them the firmware by e-mail. 

I no longer use the board myself and have switched to a Duet3d board.