makerbase-mks/MKS-TFT #168: Nozzle temperature and heated bed temperature are set to zero

Renslmf23, 2018-07-04:
Hi,

My MKS tft28 v2.0 doesn't work. At first, it would give checksum errors (error like: line number is not last line number +1, no checksum with line number XXX). I reuploaded the marlin firmware on the control board (mks v1.4), reuploaded the firmware of the MKS tft and reuploaded my G-Code. It prints normal with the usb cable, but when printing from the sd or usb it sets the nozzle temp. and bed temp. to 0, as stated in the title. It doesn't print and manually setting the temp. doesn't work, too (same issue). I've tried different flex cables, different software versions, different MKS tft screens (two v1.2's), but nothing seems to work. What am I doing wrong? 

Thanks in advance and kind regards,

Rens van Breukelen

makerbase-mks, 2018-07-16:
Have you tried printing with the Proterterface on your PC?

Renslmf23, 2018-07-16:
We have just found the solution. A marlin update updated the pins.h file, changing the pinout of the connector. This caused this weird issue.

closed 2018-07-17 by makerbase-mks