makerbase-mks/MKS-TFT #169: 3 commands freeze TFT2.4

dmitrygorbenko, 2018-07-08:
Hi,
I have TFT 2.4 with latest (up to this day) firmware.
Create a gcode-file containing three commands:
```
G28 
G1 Z50 
G1 X100 Y100
```
save to sd-card, insert it into TFT2.4, and print this file.

Expected: printer moves to specified coordinates after homing.

Actual: nothing happens, TFT2.4 panel freezes.

Any solutions ?

dmitrygorbenko, 2018-07-08:
Attaching here my config file.
[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/2173799/mks_config.txt)

makerbase-mks, 2018-07-12:
You can use the host computer to print whether the test is normal.

dmitrygorbenko, 2018-07-14:
Hi,
here is the video I recorded to show how computer executes these command on printer:
https://www.youtube.com/watch?v=wUbZ0gMWEWM

summary: octoprint on raspberry pi executes all correctly.
MKS TFT 2.4 freezes.

thanks,
Dima

makerbase-mks, 2018-07-16:
Others operation is  ok ? Like the move,preheat...

makerbase-mks, 2018-07-16:
Is the other operation normal? Like the move,preheat...

makerbase-mks, 2018-07-16:
You can give me a gcode file. Only 3 orders? For Marlin firmware, the temperature needs to reach HEATER_0_MINTEMP to print.

makerbase-mks, 2018-07-16:
The motherboard cannot use USB and TFT at the same time, because they are the same serial port, data conflicts.

dmitrygorbenko, 2018-07-20:
Sorry for delay, so answering your questions:
>> Is the other operation normal? Like the move,preheat...
Yeah, TFT 2.4 prints fine all the time. Everything is OK.
This edge case with 3 commands I found because I was trying to continue interrupted print, and I was testing if nozzle goes exactly to the same coordinates where it should be.

>> You can give me a gcode file. Only 3 orders? For Marlin firmware, the temperature needs to reach HEATER_0_MINTEMP to print.
Yeah, only 3 orders, not more -- that is all. I've set HEATER_0_MINTEMP to 5, and it's definetly not the case.

>> The motherboard cannot use USB and TFT at the same time, because they are the same serial port, data conflicts.
Agree, but I don't use them both. Like, the TFT freeze with these 3 commands happens without USB cable plugged in.

dmitrygorbenko, 2018-07-24:
Maybe you need anything else ? Some tests ?
Just want to be sure you have all you need to ensure this is an error.
Thanks,
Dima

makerbase-mks, 2018-07-26:
May be you should give me your firmware of motherboard.

dmitrygorbenko, 2018-07-28:
The firmware is Marlin 1.1.8.
Also attaching my config files
[Configuration_adv.h.txt](https://github.com/makerbase-mks/MKS-TFT/files/2238291/Configuration_adv.h.txt)
[Configuration.h.txt](https://github.com/makerbase-mks/MKS-TFT/files/2238292/Configuration.h.txt)

closed 2020-06-06 by makerbase-mks