makerbase-mks/MKS-TFT #170: TFT35 All buttons disappeared

GenaUser, 2018-07-12:
Hi,
I updated my TFT35 screen with the V1.0.1 firmware (mks_config.txt, mkstft35.bin, mks_pic, mks_font)
Now, the boot logo is very distorted and all the buttons are no longer displayed (The text is rightly displayed).
I have the same result using the configuration example.
How can I get back to the original functional display?

Thanks in advance

Operrr, 2018-07-14:
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT3.5/TFT35_V1.0.1%20Release%20file.zip

Official and exactly working firmware. 100%. Русский?

GenaUser, 2018-08-05:
The problem seems to be related to changes to the "mks_config.txt" file made on Mac OS (10.13.6). That seems corrupt sometimes the file (or folders?).
I didn't have the problem with windows, for the moment.
To be verified...

closed 2020-06-06 by makerbase-mks