makerbase-mks/MKS-TFT #171: TFT32 doesn't show current hotend temp but does print

XxAnthonyCxX, 2018-07-14:
[image]

The Hot end heats properly. i can check it in pronterface before using the TFT to print from. but it remains at Zero throughout the whole print process as seen in the provided picture. The heatbed temp shows properly. so i'm not sure what the issue may be.  I do notice that the hotend is listed as E0 on the TFT but on the MKS Sbase 1.3 config its listed as T1, could this be the issues ? do i need to change T1 to E0 ?

TFT32 is using latest firmware

makerbase-mks, 2018-07-16:
Is the actual temperature of the heating head also 0 ? What the version of the TFT32 firmware ? If this problem occurs every time you print, please send me your GCODE file.

XxAnthonyCxX, 2018-07-17:
the actual temp or as close to it as possible for the hotend is 210. i watched it on pronterface before i started printing using the TFT and turning off pronterface. it stays between 209.8 and 210.2. heat bed numbers are accurate 56/55. everything functions correctly. it's just the display that's not showing properly for the hotend.  you're saying it could be a Gcode that's not showing right ? i built the printer from scratch and installed all this stuff myself. so i may be a little off when it comes to gcode settings for print starts. i'll look it over and compare it to what my monoprice mini spits out.

XxAnthonyCxX, 2018-07-21:
All fixed. I appreciate the emails you sent Makerbase-mks, they kept me looking and working out the solution.  It turned otu it was all in my MKS Sbase config. for some reason the hotend was designated as T1 instead of just T. not sure if i did it myself, or if it was set that way as default. but as soon as i removed the 1 after the T , hotend started reading current temp properly.

closed 2018-07-21 by XxAnthonyCxX