makerbase-mks/MKS-TFT #173: Use both MKS PWC and MKS 220det

hermitcrabslab, 2018-07-19:
How I can enable/use both?
I want to use the PWC for shutdown/power-up(confirmed working with marlin 2.0x bugfix firmware but not full working with smoothieware, only power up and not shutdown)
And MKS 220 detect for ups and power loss detection to put the printer on pause until power is up again.
How to do it?

makerbase-mks, 2018-08-20:
[image]

hermitcrabslab, 2018-08-20:
I think you did not understand my question, I have both MKS PWC and MKS 220DET and I want to use both with MKS TFT32, problem is that in display configuration I can not enable both is ether MKS PWC or 220Det:
`#enable power detecting module(mks 220det:1; mks pwc:0)
>cfg_insert_det_module:0`
I have UPS connected to printer and I want to pausde the print when there is a power loss and also use the power off functions like auto off after print finish function...

hermitcrabslab, 2018-10-15:
@ksbhatt89 No, not yet

@makerbase-mks maybe I am dumb but I still cannot see how the display will send a pause to the printer when power runs out if is not configured, or you are saying that is not required to define it in config and once the PB0 pin is getting signal from mks 220det the display will send pause to the pinter?
I am right?

ksbhatt89, 2018-10-15:
I am using ups with mks 220det. Where do I need to connect  mks pwc in the block diagram shown in attachement? It is to be connected to ups or direct supply?
[image]

michail58, 2018-12-10:
здраствуйте форумчане у меня большая просьба нужна помощь связка mks tft 35 +sbase 1.3+mkc 220 det так вот в чем проблема до выключения питания печатает хорошо после выключения стол опускается Далее подаю питание на det 220 после подаю включение принтера голова хомится и приезжает к детали стол поднимается и пошла печать но филамент экструдер не выдавливает
куда копать уже не знаю перепробывал кажется все

LyaminSS, 2018-12-19:
> здраствуйте форумчане у меня большая просьба нужна помощь связка mks tft 35 +sbase 1.3+mkc 220 det так вот в чем проблема до выключения питания печатает хорошо после выключения стол опускается Далее [...]

Я так решил этот вопрос. Дешево, и сердито! Надежно
https://vk.com/@lyaminss-avtomaticheskoe-vykluchenie-3d-printera-po-zaversheniu-pecha

michail58, 2018-12-28:
обьясняю ситуацию Ставлю печатать деталь Через некоторое время мне например на ночь нужно выключить принтер Снимаю питание с mks det 220 Принтер останавливается стол опускается на 10 мм Я выключаю принтер Через некоторое время включаю принтер снимаю с паузы идет нагрев стола и сопла После нагрева оси двигаются в ноль дальше головка движется к детали стол поднимается начинается печать  но пруток не выдавливается

LyaminSS, 2018-12-28:
я пытался тоже печать возобновлять, и у меня она могла начаться на другой высоте, и сбить изделие. И я больше не стал экспериментировать. Мысль была, что я не верно файл конфигурации настроил.

michail58, 2018-12-28:
а почему не включается экструдер

LyaminSS, 2018-12-28:
за паузу отвечает TFT. Он на карту памяти делает запись места остановки.
Печать может не начаться из-за того, что не прогрелся экструдер.
Попробуй сперва прогрев сделать головы, потом возобновить печать.
Чтобы все прошло нормально - карточку вынимай, потом прогревай.
выключай питание, вставляй карточку, и включай, возобновляй

michail58, 2019-01-04:
У меня ура все заработало Только странно я перешел на другой слайсер и восстановление печати пошло После прогрева головка медленно подошла к детали и также медленно прошла внешнюю оболочку ну а дальше все как обычно

MsnSazabi, 2019-12-03:
> @ksbhatt89 No, not yet
>
> @makerbase-mks maybe I am dumb but I still cannot see how the display will send a pause to the printer when power runs out if is not configured, or you are saying that is not [...]

Was you able to solve this issue?

aokiakagome, 2019-12-15:
I made it like this.

https://youtu.be/az0qwhmqn3s

hermitcrabslab, 2019-12-25:
> I made it like this.
> 
> https://youtu.be/az0qwhmqn3s
Like how? Even spent time to read your google translated blog with no clear picture in the end...

aokiakagome, 2019-12-27:
Japanese translation is difficult.
sorry

I also have difficulty understanding the problem with translation.

[image]

What is the situation of the problem right now?

hermitcrabslab, 2019-12-27:
A simple but compleat schematic would worth more then a thousand words...

aokiakagome, 2019-12-28:
Wiring example

[image]

Start printing
↓
Power loss
↓
Timer relay
Auto power off after 3 minutes
↓
Power on
↓
Printing resume
↓
End of printing
↓
Auto power off after 3 minutes
(Add to end Gcode→G4 S180)

hermitcrabslab, 2019-12-29:
Very good, thanks\1

closed 2020-06-06 by makerbase-mks