makerbase-mks/MKS-TFT #174: My first SkinPack for MKS TFT3.5

Operrr, 2018-07-21:
Set for changing the appearance of the user interface MKS TFT3.5.

How to install:
1. Unzip the archive
2. Copy the mks_pic folder to an SD card or USB media
3. Insert the SD card or USB storage media into the MKS TFT and press the reset button

This is my first project so please do not judge severely, if you find any errors, write to email.

Download: https://1drv.ms/u/s!AicE0iKLGf_0iMYaz3-2eKsIn8LsAQ

makerbase-mks, 2018-08-20:
copy the file to the SD card

odersone, 2019-05-03:
Soeh, obviously you know somthing I do not, about the bmp files or whatever, to be able to convert a bmp successfully to bin for the TFT35. Any chanse you can help a struggeling friend out so I too can have my custom icons? 

When I try to convert a correct size bmp it is just scaled wrong when I upload to the TFT

makerbase-mks, 2019-05-04:
You can use this software to convert and set the corresponding size of the picture.

https://github.com/makerbase-mks/Software/tree/master/Image2LCD

odersone, 2019-05-04:
Any specific settings than might be important? Scan mode etc? the size I seem to be able to get from the original bin files with Makerbase image editor

closed 2020-06-06 by makerbase-mks