makerbase-mks/MKS-TFT #175: problem with MKS PWC

Operrr, 2018-07-22:
Hello! I connected the MKS PWC 2.0 to MKS TFT35 according to the datasheet, but the printer does not turn off. As in the usual mode of operation, and at that moment when the MKS TFT35 writes "Clossing Machine" on the contact PB4 there is a high level of 3.3v. What could be the problem? in the firmware?
what should be the signal level on pin PB4 when the printer is turned off?

makerbase-mks, 2018-08-20:
You need to make sure that the firmware on your motherboard is configured correctly, and you can use a voltmeter to measure the level of the lead pin you have configured to determine if it is configured correctly.

ksbhatt89, 2019-04-12:
Hi Operrr,

Could you find the solution? I have the same issue.

closed 2020-06-06 by makerbase-mks