makerbase-mks/MKS-TFT #178: MKS-TFT70 touch problem

lenain31, 2018-08-05:
Touch problem that is not accurate, unusable screen

makerbase-mks, 2018-08-06:
We will test and fix it

lenain31, 2018-08-12:
Je pense écran poubelle car non fonctionnel dommage 😢

makerbase-mks, 2018-08-20:
Can you provide some information and audio-visual materials?

makerbase-mks, 2018-09-05:
I'll handle it.

budi67, 2018-09-06:
A big thanks ;)

budi67, 2018-09-22:
Hello
Have you news ??
It's really ankward

makerbase-mks, 2018-09-27:
Has been updated to the firmware version of TFT70 v1.0.1 on github

lenain31, 2018-09-27:
Yes v1.0.1 but this is the same problem.

m050wx00, 2019-09-02:
> Touch problem that is not accurate, unusable screen

Hi there I also have the exact same problem  the screen needs calibrating is there a way to do this 
from Mike U.K.

m050wx00, 2019-09-02:
> Touch problem that is not accurate, unusable screen

Hi Lenain may ask if you managed to repair or calibrate your MKS TFT 70 touch screen

m050wx00, 2019-09-02:
> I have the same problem, it's really problematic, when you press on a place, there is two or three which react.
> It is even dangerous if you want to move the head after one print (for diverse reasons), you never know if the movement will be correct.
> 
> I have latest firmware

Hello there  may I ask if you have resolved your issue with MKS TFT70  as I have the same problem with over sensitivity of the screen  regards Mike U.K.

lenain31, 2019-09-02:
Et non malheureusement j'ai abandonné cette écran tactile qui est défectueux

neonaing, 2020-02-23:
Mine too, here is a video I posted on youtube and youku.

[Youtube](https://youtu.be/HL-OISJkcr8 )

[Youku](https://v.youku.com/v_show/id_XNDUzOTM0MTMzMg==.html?spm=a2hzp.8244740.0.0)