makerbase-mks/MKS-TFT #179: TFT35 Touch Screen. Double display of the first custom button

GenaUser, 2018-08-05:
Hello,
I've an embarrassing bug; The first custom button is displayed twice.
For their part, the functions are organized normally.
This shifts the functions relative to the buttons.
ex: button1 / function1; button1 / function2; button2 / function3 ....
In contrast, the "morefunc" buttons don't have this problem and displays normally (these buttons are only usable while printing)
Am I the only one with this problem?

makerbase-mks, 2018-08-06:
We will update the firmware as quickly as possible

GenaUser, 2018-08-06:
Thank you for your prompt response.
I wait for this update impatiently

Operrr, 2018-08-12:
You say that the firmware was updated, but in the list of downloaded firmware only version 1.0.1 which I downloaded more than a month ago.

makerbase-mks, 2018-08-20:
@just fix the bug ，no change the version

makerbase-mks, 2018-08-20:
@Operrr  just fix the bug ，no change the version

closed 2020-06-06 by makerbase-mks