makerbase-mks/MKS-TFT #181: TFT35 - Z axis crash after pause and printing resume

tiwanacote, 2018-08-09:
Hi! I want to pause printer in order to continue printing other day. 
To avoid extruder overheating, the ideal condition is to reduce temperature before switch off the machine, so you should do the routine below but printer crashes.

While printing...

1) Pause printing
2) Go to temperature control menu and turn-off extruder (You can look how temperature decrees in this menu). Here is the FIRST BUG: When return to principal screen menu don't shows temperature changes
3) Turn-off machine
4) Turn-on machine and resume printing. Here the SECOND BUG: The machine crash Z AXIS with platform

Regards

makerbase-mks, 2018-08-20:
you shouldn't reduce temperature before switch off the machine

tiwanacote, 2018-08-21:
It is not recommendable to turn off your printer before extruder has a temperature below 100°C. Thermal inertia could cause extruder overheating.

closed 2020-06-06 by makerbase-mks