makerbase-mks/MKS-TFT #185: MKS DET with MKS TFT 3.5"

Operrr, 2018-08-12:
Dear developers, if you create a device, then please solve the problems of users and not be silent. I bought MKS DET and connected it as it is written in the datasheet, but it does not work, rather the unit itself is working but the display does not react to the signals from the module. Please do not be silent and tell me the solution.

Part of my mks_config:

#enable UPS? (yes: 1; no: 0)
> cfg_have_ups_device: 1

#enable power detecting module (mks 220det: 1; mks pwc: 0)
> cfg_insert_det_module: 1

#enable auto off after print finish function (no: 0; Yes: 1)
> cfg_print_finish_close_Machine: 1

#set PB0 signal when use it for sencond nozzle (high level: 1; low level: 0)
> cfg_PB0_trigger_Level: 0

#set PB1 signal (high level: 1; low level: 0)
> cfg_PB1_trigger_Level: 0

#set error range of Z-axis on breakpoints recovery
> cfg_breakpoint_z_error: 0.2

#the function of power detecting and filament detecting (1: disable; 0: enable)
> cfg_mask_PB0_PB1_function: 0

The method of saving the gcode data with power off (1: Capacitor storage; 0: File system save)
> cfg_pwroff_save_mode: 0

makerbase-mks, 2018-08-20:
If the TFT display pause status when connected the "B0" to "-(GND)"， it means that the firmware is normal.

Operrr, 2018-08-20:
thanks, check

Operrr, 2018-09-06:
Everything worked! The problem was with the manufacturer!
On the connector, the contacts PB0 and 3.3v were interchanged. Usually the power wire is red, minus black, but in my case the positive contact was green !!! changed places and everything worked!

closed 2020-05-27 by makerbase-mks