makerbase-mks/MKS-TFT #188: sd error !!!

SHBNIK, 2018-08-18:
hi there 
i have a problem when i insert the sd to mks tft32 it shows the gcodes and every thing is ok but when i start printing a file  the boards hang and nothing work untill i restart every thing my sd is not write protected and it is 8 gig sand disk class 4 
i tested other of this sd cards and several of them have this problem !!!
what is the problem ?!
thank you for your answers !

budi67, 2018-08-19:
Same problem with a tft70, after 1,2 or several hours, the boards hang. forced to reset !!!! 
Not the first time !

makerbase-mks, 2018-08-20:
@budi67 
You can update it to the latest firmware.
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT7.0/TFT70_V1.0.1%20Release%20file.rar

makerbase-mks, 2018-08-20:
@SHBNIK  give me your email.

SHBNIK, 2018-08-20:
shahabnikkhoo@gmail.com
@makerbase-mks

closed 2020-05-27 by makerbase-mks