makerbase-mks/MKS-TFT #189: Firmware for TFT24 "Offline MKS DLC GRBL CNC Shield Controller"

DirtyEngineer, 2018-08-21:
Seems Makerbase finally released a version of the TFT24 for GRBL. (https://www.aliexpress.com/item/OFFLINE-MKS-DLC-GRBL-CNC-Shield-controller-TFT24-touch-screen-CNC-engraving-laser-control-board-DIY/32911144359.html) 

This is really exciting as I've been asking for this for a while. (https://github.com/makerbase-mks/MKS-TFT/issues/97)

Can you please provide the firmware update for the TFT24 to convert to the GRBL offline Controller.

Thank you.

makerbase-mks, 2018-09-05:
https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS_TFT_CNC

closed 2018-09-05 by DirtyEngineer

makerbase-mks, 2018-10-17:
@Donderda Sorry, for the time being only the version of tft24

blikkies, 2018-11-07:
Do you have roadmap for TFT32 CNC release?

weed2all, 2018-11-08:
Do we need to make some mods on grbl file to get mks tft24 to work...or is just plug and play?

makerbase-mks, 2018-11-13:
@weed2all As long as the baud rate of the screen is the same as the baud rate of the motherboard firmware setting, no other settings are required.

cmwwebfx, 2018-12-14:
Thank you very much for the work done on this for the CNC people. By looking at this GUI I assume this is using the red version GUI?

Could you pleas show photos of each screen? I will be looking to redesign a new GUI that will be a bit more clean and modern.

sasha-zimin-81, 2019-04-14:
TFT24 для GRBL.  Подскажите пожалуйста, как правильно залить эту прошивку для использования лазера, а то боюсь испортить драйвер.

rlwoodjr, 2020-10-01:
Is there source code available for this project?

paragraf9282, 2021-02-07:
TFT24 doesn’t work with GRBL laser if it has limit switches and Homing cycle enabled ($22=1)
Connected TFT24 not allow normally work GRBL laser over USB

marcelovx23, 2023-08-23:
> TFT24 doesn’t work with GRBL laser if it has limit switches and Homing cycle enabled ($22=1) Connected TFT24 not allow normally work GRBL laser over USB

how you know that?

paragraf9282, 2023-08-23:
> > TFT24 doesn’t work with GRBL laser if it has limit switches and Homing cycle enabled ($22=1) Connected TFT24 not allow normally work GRBL laser over USB
> 
> how you know that?

I found this out by connecting the TFT 24 to my TwoTrees engraver. I solved the problem with the USB by putting a power switch for the TFT 24. With it, I turned off the 5th line when I used the USB. But the limit switches had to be turned off…

marcelovx23, 2023-08-23:
> I found this out by connecting the TFT 24 to my TwoTrees engraver. I solved the problem with the USB by putting a power switch for the TFT 24. With it, I turned off the 5th line when I used the USB. [...]

Same thing but whit a router machine, power swich to tft24 and mks dlc v2, limit switches and Homing cycle should be diabled, when machine activate a limit swich, grbl go to alarm state and no way to remove alarm state whit tft24