makerbase-mks/MKS-TFT #190: MKS TFT32 cannot print from SD to MKS Gen-L running Marlin 1.1.9

AlwaysTinkering, 2018-08-22:
I've got an MKS Gen-L v1.0 board running Marlin 1.1.9 (the latest version), and an MKS TFT32 board running the v3.02 firmware.  

Attempting to print a file from the SD card just hangs.  The timer continues to update, but it appears nothing is sent to the printer.  Sometimes it shows strange/clearly incorrect values for the temps when this happens, like 333/70 for the bed temp.  

The same file on the same SD card prints fine in the 12864 board that is also attached.  I was also able to update the MKS TFT firmware to 3.02 using this SD card, so I know the TFT-32 can read it.  It can also show and browse folders.  I have tried several other SD cards, and several different files, with the same result.  

Other TFT32 functions, like home, move, etc. all work fine and control the printer as expected.

After the printing hang, I can press the print cancel, and the printer will then home x and y.  After that, other functions work fine.  I have tried 512mb, 2gb, and 8gb SD cards, all with the same result.

AlwaysTinkering, 2018-09-11:
To confirm this is just an issue with the final stable release of Marlin, I took the MKS TFT28 that has been working fine with Marlin 1.1.8 on my older 3D printer, and hooked it to the new Marlin 1.1.9 printer.  It failed in the same way.  It went to the printing screen and sat there like it is waiting for something from the printer that Marlin no longer sends.  Since Marlin 1.1.9 is the final stable release, it seems like it is something that should be supported by the TFT32.

makerbase-mks, 2018-09-15:
Give me your configuration file.I need to test it.

makerbase-mks, 2018-09-15:
I test the marlin 1.1.9 is ok.
Send your configuration to me. 
I need to test it.

AlwaysTinkering, 2018-09-16:
I've attached my Marlin 1.1.9 config files as well as my MKS-TFT32 config file
[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/2386583/mks_config.txt)
[Marlin119Config.zip](https://github.com/makerbase-mks/MKS-TFT/files/2386582/Marlin119Config.zip)
I'm using the very latest version of 1.1.9 from GitHub, not the earlier version of 1.1.9.

weed2all, 2018-09-22:
I think is because your mks config is under 250000 baud rate and your marlin is 115200...try to change either one to match other...

weed2all, 2018-09-22:
[mks_config.txt](https://github.com/makerbase-mks/MKS-TFT/files/2407516/mks_config.txt)
Here you have the mks config file with the correct baud rate according to your marlin firmware...try it and let us know if it worked

AlwaysTinkering, 2018-09-23:
[MarlinConfigsX5SBert.zip](https://github.com/makerbase-mks/MKS-TFT/files/2408922/MarlinConfigsX5SBert.zip)
Looks like I must have sent the wrong config, because in the config I'm using, they match.  If it were wrong, then the basic commands would not work either, and everything works fine except printing.  Also 250000 is the baud rate I'm using.  I think the config I sent was an earlier one from when I was using the melzi board that came with the printer, not the MKS Gen-L.  Regardless, attached is the config that I'm running.  
[MarlinConfigsX5SBert.zip](https://github.com/makerbase-mks/MKS-TFT/files/2408925/MarlinConfigsX5SBert.zip)

AlwaysTinkering, 2018-09-23:
Just to verify, I reflashed the printer with the baud rate at 115200 and put a config on the TFT32 with the baud rate setting at 3 and rebooted to make sure it was applied.  Home, move, etc. work, just like before, but when attempting to print, it hangs up before even setting the bed and extruder temperatures.

AlwaysTinkering, 2018-09-27:
It is not a lack of 2-way communication.  I can go to the heat button and heat up the extruder, and the display correctly shows the temperature rise as it heats.  All functions work except print.  I can also print from Simplify3D and from an SD card in the 12864 Smart Graphic controller.  I can control and flash the printer from the computer.

AlwaysTinkering, 2018-09-30:
Ok, I broke down and hooked up my logic analyzer between the TFT32 and the GEN-L.  Here is what's happening:

TFT32 sends:  N-1 M110*15
Marlin Replies:  ok N-1 P15 B3

(After approx 15 seconds)
TFT32 sends:  N-1 M110*15
Marlin Replies:  ok N-1 P15 B3

(After approx 30 seconds)
TFT32 sends:  N-1 M110*15
Marlin Replies:  ok N-1 P15 B3

At this point, I stopped the capture.

It looks like some kind of handshaking issue with the handling of setting the line number.

robustini, 2018-12-03:
The problem is not in Marlin but in the MKS-TFT firmware!

srbaroc, 2018-12-05:
And well already know some solution?

pjar82, 2019-02-20:
Yes I'm having this issue with the tft 28 and marlin 1.1.9 it shows the temp jumping from say 245 down to 233 and and up to 250 randomly has anyone fixed this issue yet

manolocs, 2019-02-22:
Hi there, adding my problem to the discussion, I Have a Ender3+MKSGENL+TFT28, it was working perfect with manual bed leveling, I installed BLTouch and made some changes via Pronterface, when I select to print the temps rise but does not print at all, message printing on the display but nothing moving. any idea what can be wrong?

RageFPV, 2019-02-25:
Bought a CR10S with the Creality V2.1 Board and an MKS TFT v3.0.3. Today I made the jump in updating whatever version of Marlin came on the printer to 1.1.9 following various guides etc. After autotuning and reuploading with new PID values etc I was ready to print. Selecting the file i wanted and confirming start the progress is shown with target bed and extruder temperatures. but the actual temps remained at 0 or whatever the last value was from Preheat/fillament loading. I could not get it to print.
I came accross this thread and now have it working thanks to AlwaysTinkering.
I needed to make a change in configuration_adv.h
Commented out ADVANCED_OK and now printer seems to be working. Bed heated up, everything homed, and extruder is heating after selecting a file to print. so as AlwaysTinkering suggested, there seems to be a feedback loop issue between the board and tft board when that ADVANCED_OK setting is enabled on marling 1.1.9?
Very confused by the whole marlin update process as my printer came with the touchscreen,  and all guides out there are focused on cr-10s with the standard LCD and swivel button.
Id like to work out how auto level and babysteps are done with the TFT.

guigle1, 2019-12-16:
Même problème d'impression qui ne démarre pas et la solution de commenter ADVANCED_OK fonctionne 👍

closed 2020-06-06 by makerbase-mks