makerbase-mks/MKS-TFT #194: Mks tft35 with sbase (no image)

yummyburger, 2018-09-01:
Hi, my tft35 doesnt show any image, just letters. Anyone know whats wrong?

The bootup mks logo can only see abit with lines all over

GenaUser, 2018-09-03:
Hi, I've the same bug when I modify the "mks_config.txt" file on MAC OS.
When I use Windows, I don't have the problem anymore.

makerbase-mks, 2018-09-05:
@yummyburger 
@GenaUser 
You can try this firmware.I have a suggest for you that you can  use the windows system to extract the files  if the firmware not possible.

makerbase-mks, 2018-09-05:
https://github.com/makerbase-mks/MKS-TFT/blob/master/MKS-TFT3.5/TFT35_V1.0.1%20Release%20file.zip

MTUpower, 2018-10-03:
Hi, all getting the same exact problem as yummyburger.  Even down to the pictures that he posted.  Have tried the undated file from Makerbase, using the example and or all combinations. Just will nt boot up with images!!!???  I am using a mac, as thats all I have.  
Ant suggestions would be very welcome.

MTUpower, 2018-10-04:
Hi maker base, still having no luck, it boots up get all the right into etc but no images for the buttons.  would appreciate helping getting this to work please.  Below is a picture of what I am looking at once boot up completed.  As previously stated appears to be exactly the same as yuhmmyburger.  But I do not have access to windows computer only MAC, so whats the suggested solution.  I have also tried your new release file V1.01, still no joy.
[image]

makerbase-mks, 2018-10-06:
@MTUpower 
We recommend that you try to use Windows to download files and unzip files, which may avoid or resolve this situation. I don't know what system you are using. This is also the case when someone used imac before.

MTUpower, 2018-10-06:
Hi Makerbase, thank you for the very unhelpful reply!! In my two previous posts I said that am using a a Mac, So to be more specific, the operating system is MacOS high Sierra, Version 10.13.6.  I also said that I don't have access to a Windows, I don't use that operating system.  So apart from going out and buying a windows based operating system, have you a solution to this issue, that affects Mac users?

makerbase-mks, 2018-10-11:
为您带来的不便实在抱歉，我们会在macos 系统进行测试，并尽快解决这个问题

makerbase-mks, 2018-10-17:
@MTUpower We have tested this issue. Because, in the mks-pic folder inside the image file, there are 135 bin files. If you use the iMac system software to decompress the firmware of tft35, in addition to 135 bin files in the mks-pic folder, there are some backups hidden. file. Therefore, the update of the picture will be disordered. We will fix it as soon as possible and update the version. Now your current solution is to set the display of those files that are hidden and delete them. After updating the firmware, it will display normally.

MTUpower, 2018-11-19:
Hi Makerbase,  I did try your suggestion of the 17th of October but no luck!!  It would help if you could show the particular hidden files to which you were referring. Have you any news as to when your update for all Mac users will be available?

MTUpower, 2018-11-29:
Good morning Makerbase, I am still having no luck getting the TFT35 screen icons.  I have tried to find your hidden files but have no idea where to look?  any news on the update for mac users?  as its frustrating to have a nice, but unusable screen.  It would help if you could issue a step by step guid to getting the screen working please.

MTUpower, 2018-12-20:
Hello Makerbase, and Merry Christmas, still unable to use my TFT35 screen, which is very disappointing, have you any news as to when you will resolve the Screen Icon issue for Mac users as reported back in September???

MTUpower, 2019-01-28:
Hi Makerbase, I find it inconceivable that you have not had the curtesy to reply to my requests for help and information since the 17th of October last year.  I am still have no luck getting the TFT35 screen to work.  I have tried the new release file, but still not working.  In-fact its even worse, all I am left with on the screen is Booting.... or Complete....   This really beyond the joke, and am disgusted by your lack of support.  Please do try to see your way to helping or a the very least replying.

stef-ladefense, 2019-01-29:
never change !
makerbase bande de cons !

omach, 2019-04-09:
Hi @MTUpower. The problem is with the hidden files (all files that begins with . (dot))
To remove them you can:

1. put the files in the SD card
2. open 'Terminal' and go to mks_pic folder in the SD card
3. if you want to view the hidden files you can list it with "ls -la"
3. execute 'rm .*'
4 . put the SD in the MKS TFT

this works for mi with a mac with MacOS Mojave.

MTUpower, 2019-04-10:
Hi OMACH,
you have just made my day, it worked!!!   Thank you very much.  Have to confess that I did also call apple support for a little advice, but I now have all the icons etc .  Now just have to swop it out and see if it works. Six months of pain, is now over!!!  For a bit anyway!!

juliaojr, 2020-04-23:
@omach you saved my TFT too! tks a lot! tks too @MTUpower

Cybersloth, 2021-01-13:
> Hi @MTUpower. The problem is with the hidden files (all files that begins with . (dot))
> To remove them you can:
> 
>     1. put the files in the SD card
> 
>     2. open 'Terminal' and go to mks_pic folder in the SD card with 'cd /Volumes/name_of_your_sd_card/mks_pic'
> 
>     3. if you want to view the hidden files you can list it with "ls -la"
> 
>     4. execute 'rm .*' to remove all hidden files in the folder
>        5 . put the SD in the MKS TFT
> 
> 
> this works for me with a mac with MacOS Mojave.

Works with Big Sur as well, thank you @omach