makerbase-mks/MKS-TFT #195: TFT32 - Customize home sequence?

magpern, 2018-09-03:
Is there a way to customize the homing sequence with tft32?
I have a mks sbase v1.3 and a tft32 and a BLTouch clone.
To home the machine, including the Z-axis it needs several steps.
First home X and Y, then raise Z, reset BLTouch, enable bltouch, move X and Y to mid of build area. Run a G30 to probe for the bed, disable bltouch.

The home all buttom in tft32 now only homes X and Y

bogaskilop, 2018-09-06:
I'm also interested in adding my own commands to normal homing function  i got davinci 1.0 with bigger stepper motor on extruder and i need  to add g28xy  then g91 g1y-20 g90 g28z g27 as a  normal g28 command

makerbase-mks, 2018-09-08:
You can add command to the "More" button.For example :G91;G1 Y20;G90;G28;

magpern, 2018-09-08:
Still better if design would allow for some flexability. Not all machines
are the same.

The "more" button is a nice extra fearure, but you cannot set a text for
the "more" button, meaning more then one button and you forget which one is
what.
So "more" button is quite useless

RainesX, 2019-02-17:
I too would realy like to add my own gcode to homing. Since smoothie doesnt like to home z with only a probe i cant use standard homing and there for it either crashes into the buildplate or doesnt work at all.
Right now i have to use leveling on auto to home z and that is just wrong.
Please add this in the config.

Andrei-Pozolotin, 2019-03-06:
See #285

makerbase-mks, 2019-03-11:
You can add command to the "More" button
#each command must be separated by semicolon ";"
>setmenu_func1:M84;

>moreitem_pic_cnt			0				#the number of "more" button 

#edit command for 1~7 "More" button
#each command must be separated by semicolon ";"
#each icon command must be less than 200 bytes
>moreitem_button1_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button2_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button3_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button4_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button5_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button6_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button7_cmd:G28 X0;G28 Y0;G28 Z0;

Set the number of button enable, then you can add the command you want to use after the button.

makerbase-mks, 2019-03-11:
You can add command to the "More" button
#set number of  "More" button 
>moreitem_pic_cnt:0

#edit command for 1~7 "More" button,each command must be separated by semicolon ";"
>moreitem_button1_cmd:G28 X0;

>moreitem_button2_cmd:G28 X0;

>moreitem_button3_cmd:G28 Y0;

>moreitem_button4_cmd:G28 Y0;

>moreitem_button5_cmd:G28 Z0;

>moreitem_button6_cmd:G28 Z0;

>moreitem_button7_cmd:G28;

#how many "More" button display on screen interface
>morefunc_cnt:0

#command setting on button1~6
>morefunc1_cmd:G28;
>morefunc2_cmd:G28;
>morefunc3_cmd:G28;
>morefunc4_cmd:G28;
>morefunc5_cmd:G28;
>morefunc6_cmd:G28;
Set the number of button enable, then you can add the command you want to use after the button.

magpern, 2019-03-11:
> You can add command to the "More" button
> #set number of "More" button
> 

The "more" button is a nice extra feature, but you cannot set a text for
the "more" button, meaning more then one button and you forget which one is
what.
Allow for a description or a text for the more button.

or just release the code to the public domain. Then the interface would be fixed and useful quite fast by others.
Now the consensus is - you can by the mks card, but the display is not good. Buy another display.