makerbase-mks/MKS-TFT #196: TFT 35

saliery999, 2018-09-03:
after 4 mks 35 with s base we have:
1) issue when you have Z max endstop and use bed leveling by 4 point from tft, when bed already at home and you choice any point, printer try move down anyway on 10mm and give damage to motor shaft, so totally they dont know that he already at home and no need to pull down the bed. maybe just add commad for auto home all axis ? or another solution.
2) lcd doesnt show remaining time, its very very helpfull thing, anyway smoothie have this option, maybe you just add ?
3) this lcd have not so good brightness, is any way to increase it by programming?

makerbase-mks, 2018-09-08:
> cfg_leveling_z_high: 10
Adjusting this configuration can solve the problem.

saliery999, 2018-09-17:
Now you upgrade the tft 35 firmwware and add auto-home after clicking on any point when manual leveling, BUT anyway they f**** go down on 10mm in any of "cfg_leveling_z_high: 10" command, i try any way  (-1, -10, 0, 0.2, 1, 2, 5) i angry because everytime damage my parts of printer. TFT still dont know that printer already at home pos and need just go up on work height.