makerbase-mks/MKS-TFT #200: No reports via WIFI

3kmch, 2018-09-09:
I connected the WIFI sucsessfully to the TFT. Sending commands from the computer to the device works without problems. :) But when I send "M501" it should report some eeprom settings like "M212...". Somehow the TFT filters out this reports. Any idea how I can grab these values?

Btw I set marlin in the config...

ZhuDaHai, 2018-09-11:
I think you want M503 to see the settings. 
M501 just reads the settings from EEPROM and makes them active. It doesn't necessarily report them.

hlgithub7430, 2018-09-27:
Hi Kurt, how did you make to send files to the esp module? I get from repetier host always an error.

3kmch, 2018-10-02:
@hlgithub7430 I did not try that by now. I just wanted to connect over wifi to list some settings. ;-)

Andrei-Pozolotin, 2019-03-06:
See #285

closed 2020-06-06 by makerbase-mks