makerbase-mks/MKS-TFT #204: TFT28 Gen V1.2 No communication

cmenchion, 2018-09-12:
I have a tft28 and 2x mks gen v1.2 boards. I cannot get any communication with my boards. If I connect via USB I am able to see the current temps but unable to control the printer. I am using Marlin(selected in the config), I've tried both 115200 and 250000 with fashing fw to printer and to the screen with updated changes.

I've switched and made new cables with same effect on both boards. Any ideas I can try?

cmenchion, 2018-09-14:
Follow up...I connected wires directly to the board and powered the screen via 12v. Same problem. I just hooked up my Re-Arm/ramps running smoothie and still same problems.

I'm at a loss other than something hardware related with the screen.

makerbase-mks, 2018-09-17:
You can confirm that the baud rate of the motherboard and the touch screen are the same.
>cfg_baud_rate:4
You are confirmed that the baud rate is consistent but still cannot communicate, it may be the touch screen is damaged. You can contact our sales staff to replace the product.

makerbase-mks, 2018-09-17:
You can confirm that the baud rate of the motherboard and the touch screen are the same.
cfg_baud_rate:4

You are confirmed that the baud rate is consistent but still cannot communicate, it may be the touch screen is damaged. You can contact our sales staff to replace the product.

closed 2020-03-12 by makerbase-mks