makerbase-mks/MKS-TFT #205: Adding M117 / M00 to the TFT Firmware

Rics3D, 2018-09-16:
This is a message specially for the developers at MKS.
To make the MKS TFT Display fully useable you have to add the Feature, that the Display can show M117 Messages. Also a window is needed after reading  a M00 in the gcode. In this window must be a button to start the programm after a M00 command . 

After this changes in the main Firmware, the display is fully usebal

makerbase-mks, 2018-09-17:
Thanks for your advice,we'll consider it.
You can make a custom design for it if you need the function.

Rics3D, 2018-09-17:
Hi, I can't.
It's a firmware thing, not a thing to configure in the .txt file.

superdbop, 2018-09-18:
I second this. We definitely need M0/M1 support (and whatever the equivalent is in non-marlin firmware.) It should include a countdown timer if applicable and a message string + "Continue" button. As for M117, it's just a simple text string on the printing status screen.

JosueAlejandro89, 2018-11-07:
It will be an awesome future to have the chance to display messages as we can set scripts in the gcode with the slicer and reported to the user to perform actions or notify status.

IrishMoss, 2019-04-25:
I agree completely.  Without the ability to output messages to the LCD, it limits how much we can interact with the interface.

scmanjarrez, 2019-05-12:
Completely agree, we are limited so much if we can't interact on error messages like filament sensor or M0 commands.

geins, 2019-07-14:
+1
Definitly needed, otherwise, the TFT stays just a gimmick and no professional replacement

mdelecate, 2019-09-18:
+1 Would love this feature, especially in conjunction with Octoprint!

DerJenigeDotInc, 2019-11-06:
+1
On my Sidewinder X1 it starts beeping when heating up the bed (from panel or from gcode) when i just send once an M117 an it does it also after an reset... Weird. Anyone else have this?

AlexZeGamer, 2021-04-14:
🆙
I would love to be able to use M117 commands on my screen, especially in combination with Octoprint !

SpankyDevil, 2022-10-04:
Has this feature been added yet?