makerbase-mks/MKS-TFT #207: Editing software and Marlin messages

Dpxmed, 2018-09-19:
I believe it has been asked in the past, but since there are probably no plans to release the firmware, are there any plans for a UI editor. Similar to the one Nextion has, rather than just an icon editor. I know we can add " more buttons, but i would like to see a software program that allows us to manage and change the UI as we need. Something where we could put whatever we would like where ever.

Personally I would like to be able to put the heater statuses on every page and put whatever buttons I would like on the "Home" screen. Also I think it would be important to beable to receive messages from marlin. I did a PID tune and when it failed, I never recieved the reason why on the TFT, unlike the normal lcd. Thanks

makerbase-mks, 2018-09-20:
The touch screen of TFT32_V3.0.2 will display tips that main board  return error message.

Andrei-Pozolotin, 2019-03-06:
See #285

closed 2020-03-12 by makerbase-mks